blackmagic-debug/blackmagic #521: [Bug]LPC1768 is not being correctly identified by the latest firmware.

jxsl13, 2019-09-17:
This version commit https://github.com/blacksphere/blackmagic/commit/9d7925792fde000f529726658561737fa3c96d1c
combined with this fix https://github.com/blacksphere/blackmagic/pull/433/files
seems to kind of work, but when the firmware is kinda being flashed incorrectly, as the incorrect LED blinks when flashing the Mbed(framework) Blinky example.

jxsl13, 2019-09-17:
Also why does the board have such a behavior that if you connect it to the USB port of your computer first and then to the micro controller board, it doesn't work properly, but if you connect it to the board first and then to your computer's USB port, it does work properly??!?

jxsl13, 2019-09-19:
This commit introduced the incorrectly identification of LPC17xx Microcontroller.
https://github.com/blacksphere/blackmagic/commit/302ff20a6d5b806c09e0ca7e996beab3ef3596f4

jxsl13, 2019-09-19:
It identifies the cup correctly, but fails at debugging flashed firmware. I get a SIGLOST error.

```
Reading symbols from /Users/john/Documents/PlatformIO/Projects/Testing_MCB1768/.pio/build/lpc1768/firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = blackmagic
PlatformIO: Initializing remote target...
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC17xx
0x00000004 in __isr_vector ()
Temporary breakpoint 1 at 0x236c: file src/main.cpp, line 6.
Loading section .text, size 0x804c lma 0x0
Loading section .ARM.exidx, size 0x8 lma 0x804c
Loading section .data, size 0xad0 lma 0x8054
Loading section .heap, size 0x6120 lma 0x8e50
Start address 0x173c, load size 60484
Transfer rate: 32 KB/sec, 945 bytes/write.
Warning: the current language does not match this frame.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.

Program terminated with signal 
```

jxsl13, 2019-09-19:
Reversing the commit doesn't seem to solve the problem, as debugging won't work. It seems like the checksum is missing in the .text region.

```
(gdb) target extended-remote /dev/cu.usbmodem79AC68971
Already connected to a remote target.  Disconnect? (y or n) y
Remote debugging using /dev/cu.usbmodem79AC68971
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC17xx M3/M4
(gdb) attach 1
Attaching to program: /Users/john/Documents/PlatformIO/Projects/Testing_MCB1768/.pio/build/lpc1768/firmware.elf, Remote target
0x00000004 in __isr_vector ()
(gdb) file ~/Documents/PlatformIO/Projects/Testing_MCB1768/.pio/build/lpc1768/firmware.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Load new symbol table from "~/Documents/PlatformIO/Projects/Testing_MCB1768/.pio/build/lpc1768/firmware.elf"? (y or n) y
Reading symbols from ~/Documents/PlatformIO/Projects/Testing_MCB1768/.pio/build/lpc1768/firmware.elf...done.
(gdb) load
`/Users/john/Documents/PlatformIO/Projects/Testing_MCB1768/.pio/build/lpc1768/firmware.elf' has changed; re-reading symbols.
Loading section .text, size 0x804c lma 0x0
Loading section .ARM.exidx, size 0x8 lma 0x804c
Loading section .data, size 0xad0 lma 0x8054
Loading section .heap, size 0x6120 lma 0x8e50
Start address 0x173c, load size 60484
Transfer rate: 31 KB/sec, 945 bytes/write.
(gdb) start
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Temporary breakpoint 5 at 0x236c: file src/main.cpp, line 6.
Starting program: /Users/john/Documents/PlatformIO/Projects/Testing_MCB1768/.pio/build/lpc1768/firmware.elf
Running the default executable on the remote target failed; try "set remote exec-file"?
```

UweBonnes, 2019-09-19:
I do not see any BMP specific problem here. The target is recognized, you can program flash. Keep in mind that LPC has some place where a checksum or some other magic number is stored. Siglost happens, when your target does special things, perhaps remap SWD pins or go to deep sleep.

jxsl13, 2019-09-19:
The target is recognized by a commut that's half year old. The current latest madter branch does not recognize the lpc1768 properly. it recognizes some nordic nf52 m3/m4 or so.

UweBonnes, 2020-12-09:
Please retry with current master

esden, 2021-12-15:
@jxsl13 I am closing this issue now, as I consider it fixed. If you retest and still have a problem please feel free to reopen this issue.

closed 2021-12-15 by esden

jxsl13, 2022-01-01:
well, I cannot test this anymore.
I did works with that micro controller for my bachelor's thesis.
And I'm not working with those anymore.