blackmagic-debug/blackmagic #522: Regression: NRF52832 Access Port doesn't work anymore

maz3max, 2019-09-17:
I am trying to flash a NRF52832 with activated APPROTECT.
That works with BMP Version v1.6.1-311-gfbf1963, but not with the current version.
With the current version, the BMP crashes.

maz3max, 2019-09-17:
I got following output after attaching to the debug port:
```
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
Ignoring packet error, continuing...
Reply contains invalid hex digit 116
```
After that, the BMP does not respond anymore.

maz3max, 2019-09-18:
I did some disection:
`19a8078f9934c99aaf1b88af4d41536d14c4d4b0 is the first bad commit`

maz3max, 2019-09-18:
correction: I did some bisecting and this is the result:
`7bcf7f492443d077e35d4f39dcf2c04e7594befb is the first bad commit`

maz3max, 2019-09-18:
when removing the support for "P" and "p" packets, it works again.

UweBonnes, 2019-09-18:
i will have a look next week

UweBonnes, 2019-09-23:
Fixed by #524

PR #524 (2019-09-23): Target: Default to nop-function() for all exported target functions.
Fixes #522.
commit fae2966b (2019-09-23): Target: Default to nop-function() for all exported target functions.

Fixes #522.
closed 2019-09-23 by UweBonnes