blackmagic-debug/blackmagic #523: USB not working in DFU on Blue Pill host

UberLambda, 2019-09-20:
I have a STM32F103C8T6 board ("blue pill") that used to work fine with BMP (not sure about the exact version, but it was from ~5/6 months ago).
I pulled the latest master today, did `make PROBE_HOST=stlink` and updated the board via dfu-util. This caused the blue pill not to be recognized anymore via usb; nothing happens when the device is plugged in, and when I press reset I get this output:
<details>
<summary>dmesg</summary>

```
[ 7025.713455] usb 1-1: new full-speed USB device number 67 using xhci_hcd
[ 7025.836805] usb 1-1: device descriptor read/64, error -71
[ 7026.066815] usb 1-1: device descriptor read/64, error -71
[ 7026.173495] usb usb1-port1: attempt power cycle
[ 7026.816792] usb 1-1: new full-speed USB device number 68 using xhci_hcd
[ 7026.940057] usb 1-1: device descriptor read/64, error -71
[ 7027.170164] usb 1-1: device descriptor read/64, error -71
[ 7027.400129] usb 1-1: new full-speed USB device number 69 using xhci_hcd
[ 7027.523374] usb 1-1: device descriptor read/64, error -71
[ 7027.753497] usb 1-1: device descriptor read/64, error -71
[ 7027.860170] usb usb1-port1: unable to enumerate USB device
```
</details>
If I compile and flash [this example](https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f1/other/usb_hid) to the board it works fine. dfu_util.bin, however (flashed via serial port & stm32loader) does not - same dmesg output as above.

I also tried patching `stlink/platform.c` with the code at https://lujji.github.io/blog/installing-blackmagic-via-stlink-bootloader/ (as my blue pill seems to have the 10kOhm R10 problem) but the problem persists.

UweBonnes, 2019-09-21:
Use the swlink platform for bluepill. check the readme´s again.

UberLambda, 2019-09-21:
Hello, tried `make clean && make PROBE_HOST=swlink` but I have almost the same problem after flashing:
<details>
<summary>dmesg</summary>

```
[  874.394035] usb 1-1: new full-speed USB device number 30 using xhci_hcd
[  874.517375] usb 1-1: device descriptor read/64, error -71
[  874.747384] usb 1-1: device descriptor read/64, error -71
[  874.977353] usb 1-1: new full-speed USB device number 31 using xhci_hcd
[  875.100704] usb 1-1: device descriptor read/64, error -71
[  875.330704] usb 1-1: device descriptor read/64, error -71
[  875.437439] usb usb1-port1: attempt power cycle
[  876.080698] usb 1-1: new full-speed USB device number 32 using xhci_hcd
[  876.080981] usb 1-1: Device not responding to setup address.
[  876.287568] usb 1-1: Device not responding to setup address.
[  876.494038] usb 1-1: device not accepting address 32, error -71
[  876.617353] usb 1-1: new full-speed USB device number 33 using xhci_hcd
[  876.617543] usb 1-1: Device not responding to setup address.
[  876.824205] usb 1-1: Device not responding to setup address.
[  877.030694] usb 1-1: device not accepting address 33, error -71
[  877.030806] usb usb1-port1: unable to enumerate USB device
```
</details>
BMP is not detected neither by lsusb nor dfu-util.
I also tried `$ echo Y | sudo tee /sys/module/usbcore/parameters/old_scheme_first` but it didn't improve things.

UweBonnes, 2019-09-21:
If you have flashed the wrong firmware, you must force bootloader entry, Read the readmes how to do so, there is no way to do so via USB when USB is not working.. The readmes also deprecated dfu-util and probably mention nowhere stm32loader.Read also about STM32F103x8 and the code size of BMP.

And please, do not edit the message a zillion times, think more carefully before writing., The edits clutters the BMP gitter log.

UberLambda, 2019-09-22:
I have found [what I think is the relevant README](https://github.com/blacksphere/blackmagic/blob/master/src/platforms/swlink/README.md) but I still haven't managed to make BMP work.
My blue pill has 128kb flash so the new firmware should work fine on it.
`make clean && make PROBE_HOST=swlink`
1. BOOT0 set, flash blackmagic_dfu.bin via serial, unset BOOT0, set BOOT1, reset -> device descriptor error
2. BOOT0 set, flash blackmagic_dfu.bin via serial, unset BOOT0, reset -> device descriptor error
2. STLINK/V2 -> write blackmagic_dfu.bin to 0x08000000, set BOOT1, reset -> device descriptor error
2. STLINK/V2 -> write blackmagic_dfu.bin to 0x08000000, write blackmagic.bin to 0x08002000, reset -> device descriptor error

BMP used to work fine on the same exact blue pill; I would gladly test an older version but sadly stm32duino.com is not accessible anymore (so neither are the precompiled binaries that were there) and compilation fails on older (~6 months ago) commits on my machine (
<details>
<summary>GCC output</summary>

```
target/efm32.c:61:17: error: cast between incompatible function types from '_Bool (*)(target *)' {aka '_Bool (*)(struct target_s *)'} to '_Bool (*)(target *, int,  const char **)' {aka '_Bool (*)(struct target_s *, int,  const char **)'} [-Werror=cast-function-type]
   61 |  {"erase_mass", (cmd_handler)efm32_cmd_erase_all, "Erase entire flash memory"},
      |                 ^
target/efm32.c:62:13: error: cast between incompatible function types from '_Bool (*)(target *)' {aka '_Bool (*)(struct target_s *)'} to '_Bool (*)(target *, int,  const char **)' {aka '_Bool (*)(struct target_s *, int,  const char **)'} [-Werror=cast-function-type]
   62 |  {"serial", (cmd_handler)efm32_cmd_serial, "Prints unique number"},
      |             ^
target/efm32.c:63:15: error: cast between incompatible function types from '_Bool (*)(target *)' {aka '_Bool (*)(struct target_s *)'} to '_Bool (*)(target *, int,  const char **)' {aka '_Bool (*)(struct target_s *, int,  const char **)'} [-Werror=cast-function-type]
   63 |  {"efm_info", (cmd_handler)efm32_cmd_efm_info, "Prints information about the device"},
      |               ^
```
)

UberLambda, 2019-09-22:
I compiled an older commit of BMP (0793dac2cf901ff44c94316cca6ddd1a59648c0d, disabled -Werror) and DFU works on a different, 64kB blue pill (PROBE_HOST=stlink). The same binary, however, does not on the 128kB blue pill - but [this USB HID example](https://github.com/libopencm3/libopencm3-example/tree/master/examples/stm32/f1/ther/usb_hid) does...

whitequark, 2019-09-22:
> And please, do not edit the message a zillion times, think more carefully before writing., The edits clutters the BMP gitter log.

You should simply fix you tooling instead of being rude to contributors.

UberLambda, 2019-10-14:
I managed to source another 128k blue pill and the latest firmware is working on it, so I assume the USB connector has failed somehow. It is not clear why it works with the USB example above, but I assume the issue was on my end, sorry.

closed 2019-10-14 by UberLambda