blackmagic-debug/blackmagic #526: Feature Request: Unified way to unlock chip protection

maz3max, 2019-09-25:
Unlocking the read-back protection, access-port protection etc. seems to be different on different chips.
On a NRF52 chip, I can unlock the chip by attaching to the `Access Port` and issuing `monitor erase_mass`.
On an STM32F1 chip, I would have to attach to the usual debug port, issue `monitor option erase` and power-cycle.

What about a `monitor unlock` command?
Or maybe an option like `auto-unlock`?

UweBonnes, 2019-09-25:
Nobody has all involved chips. So as long as the procedure is done chip specific, those who have the chip can contribute and test. With a generic procedure, the one implementing this generic approach would need to implement  and test for all chips in a way that handles all the specificas of the chips in a sensible API. So I guess nobody will come up with such a feature.

maz3max, 2019-09-25:
I would suggest a command that is always available.
It could bundle the already existing (!) ways to unlock specific chips.
When there is no such procedure implemented for the specific chip, an error message could be generated.
There is absolutely no need for a single person to test every chip. I would be happy to test it for stm32f1, nrf51 and nrf52 chips.

esden, 2021-12-15:
I think I can agree that it would be nice to consolidate the different commands used to unlock the chips. But I think that we should not have the command available when it is not implemented. All we need for that is have a command hook and not show the command in the help when it is not available.

@maz3max If you are still interested feel free to take a stab at a patch to implement that.

closed 2021-12-16 by maz3max