blackmagic-debug/blackmagic #532: nRF52840 not fully detected?

sjlongland, 2019-10-11:
Hi all,

We've just purchased a Black Magic Probe to program some Fanstel BT840X dongles (which are built around the nRF52840)… for some reason the chip isn't being detected fully and doesn't seem to be able to load code properly.

```
stuartl@vk4msl-ws:~/vrt/projects/widesky/hub/openthread$ gdb-multiarch output/nrf52840/bin/ot-cli-ftd 
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from output/nrf52840/bin/ot-cli-ftd...done.
(gdb) target extended-remote /dev/ttyACM1
Remote debugging using /dev/ttyACM1
(gdb) monitor version
Black Magic Probe (Firmware v1.6.1-1-g74af1f5) (Hardware Version 3)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      ARM Cortex-M
(gdb) attach 1
Attaching to program: /home/stuartl/vrt/projects/widesky/hub/openthread/output/nrf52840/bin/ot-cli-ftd, Remote target
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0x00015fd6 in ot::Mle::MleRouter::HandlePartitionChange (this=0x200000c0 <object+4>) at /home/stuartl/vrt/projects/widesky/hub/openthread/examples/../src/core/thread/mle_router.cpp:98
98      }
(gdb) monitor mass_erase
Target does not support this command.
(gdb) monitor erase_mass
Target does not support this command.
(gdb) load
Loading section .text, size 0x43174 lma 0x0
Loading section .ARM.exidx, size 0x8 lma 0x43174
Loading section .data, size 0xb8 lma 0x4317c
Start address 0x27f08, load size 274996
Transfer rate: 50 KB/sec, 978 bytes/write.
(gdb) compare-sections
Section .text, range 0x0 -- 0x43174: MIS-MATCHED!
Section .ARM.exidx, range 0x43174 -- 0x4317c: MIS-MATCHED!
Section .data, range 0x4317c -- 0x43234: MIS-MATCHED!
warning: One or more sections of the target image do not match
the loaded file

(gdb) 
```

I'll admit I haven't done much with this tool but the one I bought some months ago for my home use seemed to talk to the nRF52840 (I could be mistaken though).  The one my workplace bought a few weeks ago (on my recommendation) doesn't seem to.

Is there some step I missed?

sjlongland, 2019-10-11:
Okay, today's latest gets _further_… not quite there yet though:
```
stuartl@vk4msl-ws:~/vrt/projects/widesky/hub/openthread$ gdb-multiarch output/nrf52840/bin/ot-cli-ftd 
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from output/nrf52840/bin/ot-cli-ftd...done.
(gdb) target extended-remote /dev/ttyACM0
/dev/ttyACM0: Device or resource busy.
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Nordic nRF52 M3/M4
 2      Nordic nRF52 Access Port 
(gdb) attach 1
Attaching to program: /home/stuartl/vrt/projects/widesky/hub/openthread/output/nrf52840/bin/ot-cli-ftd, Remote target
0x00015fd6 in ot::Mle::MleRouter::HandlePartitionChange (this=0x200000c0 <object+4>) at /home/stuartl/vrt/projects/widesky/hub/openthread/examples/../src/core/thread/mle_router.cpp:98
98      }
(gdb) load
Error erasing flash with vFlashErase packet
(gdb) load
Error erasing flash with vFlashErase packet
(gdb) monitor help
General commands:
        version -- Display firmware version info
        help -- Display help for monitor commands
        jtag_scan -- Scan JTAG chain for devices
        swdp_scan -- Scan SW-DP for devices
        targets -- Display list of available targets
        morse -- Display morse error message
        halt_timeout -- Timeout (ms) to wait until Cortex-M is halted: (Default 2000)
        connect_srst -- Configure connect under SRST: (enable|disable)
        hard_srst -- Force a pulse on the hard SRST line - disconnects target
        tpwr -- Supplies power to the target: (enable|disable)
        traceswo -- Start trace capture [(baudrate) for async swo]
ARM Cortex-M specific commands:
        vector_catch -- Catch exception vectors
nRF52 specific commands:
        erase_mass -- Erase entire flash memory
        read -- Read device parameters
(gdb) monitor erase_mass
erase..
(gdb) load
Error erasing flash with vFlashErase packet
```

UweBonnes, 2019-10-11:
Is the NRF IO Voltage connected to the BMP? Erase_mass will switch to 1.8 Volt VCC, if the BMP buffers do not see that voltage, they loose the connection.

I had success with the NRF52 dongle and a native BMP.

sjlongland, 2019-10-11:
On 11/10/19 7:54 pm, UweBonnes wrote:
> Is the NRF IO Voltage connected to the BMP? Erase_mass will switch to
> 1.8 Volt VCC, if the BMP buffers do not see that voltage, they loose the
> connection.
> 
> I had success with the NRF52 dongle and a native BMP.

That I'm not certain of, Fanstel don't provide schematics it seems, and
the chip is buried under a big RF shield.  There's just a tag-connect
footprint to link to the BMP.

The board is being powered from USB at the time of doing the mass erase.
  Is the 1.8V power rail a requirement of the BMP for mass erase?
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

UweBonnes, 2019-10-11:
Measure the IO Voltage on the BMP after mass erase. If it is 3.3 Volt, the error is clear. You need to provide TPWR with 1.8 Volt.

sjlongland, 2019-10-11:
Right, I'll see if I can dig up the details on the tag connect footprints and locate the `TPWR` pad on Monday for some measurements.

closed 2019-10-16 by sjlongland

UweBonnes, 2019-10-16:
Having insufficent debug connections is not a problem of the BMP probe.

Please test again, only flashing the code without full erase and without touching the fuses. That way the IO Voltage is kept and things should work. And it would also work if you probe is 1.8 Volt compatible and the NRF IO-Voltage is connected to the BMP probe.