blackmagic-debug/blackmagic #537: STM32H7 not fully detected using jtag_scan

silvergasp, 2019-10-14:
I have been attempting to program an stm32h7 using both jtag and swd. When using swdp_scan the stm32h7 driver is selected correctly. However using jtag_scan the stm32h7 driver is not selected.
```
(gdb) monitor jtag_scan
Target voltage: 3.4V
Available Targets:
No. Att Driver
 1      STM32L4Rx M3/M4
 2      ARM Cortex-M M7
(gdb) 
```
Note that in this case the stm32h7 and the stm32l4r are connected in a daisy Daisy-chained JTAG (IEEE 1149.1). Where the stm32h7 is the first in the chain and the stm32l4 is the second in the chain.

In this case the stm32h7 has been detected as a generic Cortex-M7 processor. This is correct and allows the first 1MB flash bank to be programmed without any issues. However binaries greater than 1MB cannot be flashed as the second flash bank is not recognised.

I would be happy to work on this. I am new to this project, so some guidance on how to approach this issue would be appreciated.

silvergasp, 2019-10-14:
Bit of an update, I worked out how to compile a debug build. This is the output with debug_bmp on /dev/ttyACM1 whilst running jtag_scan

Unref AP
Unref AP
/Idle
Return to Run-Test/Idle
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040
ROM: Table BASE=0xe00ff000 SYSMEM=0x1
0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0xlx)-> cortexm_probe
1 0xe0001000: 0x00000000 <- does not match preamble (0xB105000D)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0xlx)
3 0xe0000000: 0xb1b1b1b1 <- does not match preamble (0xB105000D)
4 0xe0040000: 0x00000000 <- does not match preamble (0xB105000D)
5 0xe0041000: Debug component - Cortex-M4 ETM (Embedded Trace) (PIDR = 0xlx)
ROM: Table END
AP   0: IDR=84770001 CFG=00000000 BASE=e00fe003 CSW=43800040
ROM: Table BASE=0xe00fe000 SYSMEM=0x1
ROM: Table BASE=0xe00ff000 SYSMEM=0x1
 0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0xlx) -> cortexm_probe
LPC11/84: IDCODE 0x00000000
 1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0xlx)
 2 0xe0002000: Generic IP component - Cortex-M7 FBP (Flash Patch and Breakpoint) (PIDR = 0xlx)
 3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0xlx)
 4 Entry 0xfff41002 -> Not present
 5 Entry 0xfff42002 -> Not present
 ROM: Table END
1 0xe0041000: Debug component - Cortex-M7 ETM (Embedded Trace) (PIDR = 0xlx)
2 0xe0043000: Debug component - CoreSight CTI (Cross Trigger) (PIDR = 0xlx)
3 Entry 0x1ff02002 -> Not present
ROM: Table END
AP   1: IDR=84770001 CFG=00000000 BASE=00000002 CSW=43800040
AP   2: IDR=54770002 CFG=00000000 BASE=e00e0003 CSW=80000040
ROM: Table BASE=0xe00e0000 SYSMEM=0x0
0 Entry 0x1002 -> Not present
1 Entry 0x2002 -> Not present
2 0xe00e3000: 0x00000000 <- does not match preamble (0xB105000D)
3 0xe00e4000: 0x00000000 <- does not match preamble (0xB105000D)
4 0xe00e5000: PrimeCell peripheral - System TSGEN (Time Stamp Generator) (PIDR = 0xlx)
 Fault reading ID registers
ROM: Table END

UweBonnes, 2019-10-14:
What STM32H7? Single core H743 or a dual core >= H745.  Firmware using WFI or not?

You can now also get the debug output more easily:
- Compile recent firmware and flash
- Compile for "PROBE_HOST=pc-hosted" and start "blackmagic_hosted -s /dev/<whateverACM>" and connect gdb to :2000.
- Now you can see debug messages on the starting console and changing anything beside the basic JTAG/SWD/device setting command can be done on the PC side without debugging.

Correct probing is a tediuos subject. #507 tries to rework probe/attach. Especially STM32 should get better with it. E.g. protected SAMD21 shows other errors, but I now have  a device to test.

Please try #507 abd report.

UweBonnes, 2019-10-14:
In general, few people use full JTAG programming and even less fewer JTAG chains as BMP targets. So few work has been done on that. Debug and Patches welcome!

silvergasp, 2019-10-15:
> What STM32H7? Single core H743 or a dual core >= H745. Firmware using WFI or not?

We are using the stm32h753, this is a single core model of the stm32h7. Note that the H753 is identical to the H743, with the addition of a encryption coproccesor. They share the same reference manual [RM0433](https://www.st.com/content/ccc/resource/technical/document/reference_manual/group0/c9/a3/76/fa/55/46/45/fa/DM00314099/files/DM00314099.pdf/jcr:content/translations/en.DM00314099.pdf). 

Also note that the stm32h753 shows up as an stm32h743 under swd. This seems to be a side effect of them sharing the same jtag ID. Looking into the source code for probing I don't think this is an issue, however I will update the message to match the "STM32H742, STM32H743/753 and STM32H750"

WFI is not used at this point. However it is my intention to use it in future.

I will try the PC hosted version of GDB and report back.

silvergasp, 2019-10-15:
Do you know of anyone that has attempted to use jtag chaining? 

It would be a good place to start to see combinations of chips that have worked so I can cross compare against my setup using an stm32l4 and a stm32h7 in a jtag chain.

UweBonnes, 2019-10-20:
Stm32[F|H]7 uses ADI_DPV2, but only for the SWDP. It also has a lot of AP. I am able to see the H7 with #507, that shuffles around a lot. Perhaps give it a try abd give feedback.

Enzeys, 2019-11-17:
Hello, I'm sorry I've nothing to contribute at the moment; but do either of you know wether or not it's possible for me to program and possibly debug [STM32F722RET6](https://datasheet.lcsc.com/szlcsc/1810010121_STMicroelectronics-STM32F722RET6_C118207.pdf) through SWDP? The device itself does support SWD, but it's a Cortex-M7, that's why I'm asking.

UweBonnes, 2019-11-17:
It is bad habit to highjack some unrelated thread.

Did you try to connect? BMP should be able!

Enzeys, 2019-11-17:
I realise that, I'm really sorry, I guess I should have made my own 'issue.' even if it is just a question and not an actual issue?
I don't have the device, I'm planning to use it in my next project and just needed to know if I could program it through BMP. I've never changed BMP code before so I'm not sure how difficult it would be to add the device I want.

UweBonnes, 2019-11-17:
The F722 is added since long. but I do not know if the BMPV2.1 release already includes F722.

Enzeys, 2019-11-17:
I'm using a bluepill as BMP with the latest commit so I think it's good then! Thank you :) I really appreciate your help. And once again, I'm sorry for hijacking the thread.

UweBonnes, 2020-12-09:
Please try again with current git head

esden, 2021-12-15:
Considering that this issue is now quite stale. I will assume that this issue is fixed. If I am wrong about this please feel free to reopen this issue.

closed 2021-12-15 by esden