blackmagic-debug/blackmagic #538: BMP serial interface ignores write (PC->device) bigger 16 byte.

UweBonnes, 2019-10-14:
When writing blocks of > 16 bytes without delay between character, BMP ignores the whole block.

zyp, 2020-04-30:
I just ran into this issue myself. Issue is present both in 2b4000b which I had loaded on my BMP and 08bb4d5 which is current master.

Looking at captures for the USB traffic, the device returns a STALL handshake to large packets: 
[1](https://bin.jvnv.net/file/GtKiI.png) [2](https://bin.jvnv.net/file/8GyeA.png)

First capture is from actual use, where I first spotted the problem. Second capture is reproducing the problem by simply doing `cat /dev/zero > /dev/…`.

There's not really any reasonable reason for why the device should return a STALL in this situation and I don't see anything suspicious in `usbuart.c` either, so I suspect this might be something that will have to be fixed in libopencm3.

zyp, 2020-04-30:
Found the problem. 3e3e450 reduced the size of the buffer for the endpoint without also reducing the max packet size in the endpoint descriptor, so hardware only expects 32B packets while telling the host it can send 64B packets, and when receiving a larger packet than expected, the STALL handshake signals that error.

PR #652 (2020-05-01): cdcacm: Fix UART interface OUT endpoint MPS in descriptor.
3e3e450 reduced the buffer size for the UART interface OUT endpoint to
32B, but left wMaxPacketSize at 64B. This effectively configures the
hardware to receive packets up to 32B in size while telling the host it
can send larger packets. On reception of a larger packet, the usb core
treats it as an error, drops the packet and replies with a STALL
handshake.

This commit fixes the issue by updating the descriptor to match the
buffer configuration.

Fixes #538
commit d63e870e (2020-05-01): cdcacm: Fix UART interface OUT endpoint MPS in descriptor.

3e3e450 reduced the buffer size for the UART interface OUT endpoint to
32B, but left wMaxPacketSize at 64B. This effectively configures the
hardware to receive packets up to 32B in size while telling the host it
can send larger packets. On reception of a larger packet, the usb core
treats it as an error, drops the packet and replies with a STALL
handshake.

This commit fixes the issue by updating the descriptor to match the
buffer configuration.

Fixes #538
closed 2020-05-01 by UweBonnes