blackmagic-debug/blackmagic #541: Command syntax for "mon jtag_scan <list of IR lengths>" not documented

UweBonnes, 2019-10-19:
Running "mon jtag_scan help" does not list the possible command options. I did not find an example in the wiki neither.

ruabmbua, 2019-10-19:
I think you are supposed to add a list of integers, which are the IR-lens of the peers in the JTAG daisy chain structure. When they are not specified, the BMP will try to probe for the IR-lens.

esden, 2021-12-15:
@UweBonnes you are right. Do you want to add the '<list of IR lengths>' to the help output?

esden, 2022-08-02:
As we don't really have much flash space for the more advanced function use I think this should be documented on the https://black-magic.org website. For that we should create a dedicated man page style section on the website. I am thus transferring this issue to the website repo.