blackmagic-debug/blackmagic #544: Black Magic Settings for IAR

juttTechnology, 2019-10-19:
hi,
   I like to use the BMP with IAR. I am using bluepill, successfully flashed with the latest release and com ports also visible in device manager. I am using windows 10. is there any special setting for IAR compiler ?

esden, 2021-12-15:
I don't think BMP is supported by the IAR toolchain. We do not currently have plans to add support for it. You can still simply start GDB on your command line, flash and debug the artifacts generated by IAR.

All this considered, I will close this issue for now. When someone with an actual plan and/or knowledge of IAR wants to contribute support and documentation feel free to reopen this issue or create a new one to track progress.

dragonmux, 2023-12-12:
What happens if you put in `\\.\COMn\` where the lower-case n is the serial port number for the GDB interface on BMP? Unless they're being silly about it, that should work we'd think.

Regarding proxying, if using BMP properly doesn't wind up working, build BMDA (`make PROBE_HOST=hosted HOSTED_BMP_ONLY=1`) via MSYS2's UCRT64 environment. BMDA spins up an interface on localhost:2000 - no need to write something new to make this work in the advent that they've done a silly and not let you put arbitrary text in the box.

dragonmux, 2023-12-13:
It wouldn't be any easier with CMake and there is not a prebuilt binary for now due to dependencies - BMDA on Windows uses the MSYS2 UCRT64 environment and there is a complete tested guide scheduled for the website: https://github.com/blackmagic-debug/black-magic-org/commit/d51817e17c90efb2f0c4898f5b5ed241c4350e12 - please give this a try. Iff, because you're on W7, the UCRT64 environment fails to run and work, you will need to instead use the MinGW64 environment.

The instructions have been written with `pacboy` so should gracefully work on the new environment, you just won't be able to run the resulting binaries outside of it (and now you see why we don't have prebuilt binaries for now).