blackmagic-debug/blackmagic #547: Support for stm32L071CZ

JuttDB, 2019-10-21:
Hi,
   I am using bluepill board and its working great, I can erase stm32f401re but if I try to erase the stm32L071cz then I received an error like this option does not support.

UweBonnes, 2019-10-21:
"I received an error like this option does not support." 

Quote the exact error!

JuttDB, 2019-10-21:
C:\Users\Desktop>arm-none-eabi-gdb -ex "target extended-remote \\.\COM13"
C:\Program Files (x86)\GNU Tools ARM Embedded\8 2019-q3-update\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GNU gdb (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Remote debugging using \\.\COM13
(gdb) monitor swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32L0x M0+
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x08000886 in ?? ()
(gdb) mon erase_mass
Target does not support this command.

UweBonnes, 2019-10-21:
From the source:
   o Mass erase unimplemented.  The method for performing a mass erase
     is to set the options for read protection, reload the option
     bytes, set options for no protection, and then reload the option
     bytes again.  The command fails because we lose contact with the
     target when we perform the option byte reload.  For the time
     being, the command is disabled.

JuttDB, 2019-10-21:
thank you for the detailed answer. so let this ticket open until there is no workaround

esden, 2021-12-15:
@UweBonnes just checking. Is this mass_erase still not implemented?

dragonmux, 2022-07-15:
It was, and this is now fixed in #1108

closed 2023-04-03 by esden