blackmagic-debug/blackmagic #548: DFU binaries - automated builds

Adminius, 2019-10-28:
Hi, 
as described in [readme.md](https://github.com/blacksphere/blackmagic/blob/master/README.md) users can use pre-build binaries from here: http://builds.blacksphere.co.nz/blackmagic
But there ist no DFU files available. Is it possible to add DFU as well?
For (Windows) users it is a lot easier to use pre-build binaries (DFU+FW) instead of installing cygwin or compile binaries by it self.

UweBonnes, 2019-10-28:
What are "DFU binaries"? You can already upload the binaries via DFU, e.g. dfu-utils.

Adminius, 2019-10-28:
This dfu-binaries:
https://github.com/blacksphere/blackmagic/wiki/Hacking

blackmagic_dfu - ELF binary of the Black Magic DFU bootloader.
blackmagic_dfu.bin - Flat binary of the DFU bootloader, load at 0x8000000.

UweBonnes, 2019-10-28:
DFU files are only needed for initial bringup. No need to update otherwise.

commit 04d9749f (2019-10-28): Makefile: Publish DFU binaries with the daily build. (#549)

Fixes #548.
closed 2019-10-28 by UweBonnes