blackmagic-debug/blackmagic #553: Support for LPC546xx

itavero, 2019-11-15:
I'd like to see support for the [NXP LPC546xx processor family](https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/general-purpose-mcus/lpc54000-cortex-m4-/power-efficient-microcontrollers-mcus-with-advanced-peripherals-based-on-arm-cortex-m4-core:LPC546XX) in the Black Magic Probe.

UweBonnes, 2019-11-15:
Sorry, that is not how BMP works. BMP needs contributions.

You probably have the chip and the docs. Try to add something yourself, ask here or on discord if you find problems.
- Is the chips similar to some other LPC? Then maybe only an additional entry is needed.

itavero, 2019-11-18:
Hi @UweBonnes, it's not that I did not want to do this myself. I just did not yet find the time to do so, so I figured open up an issue, so someone else might be able to beat me to it ;)

closed 2020-09-30 by UweBonnes