blackmagic-debug/blackmagic #557: Wrong Register values used in NRF52 MDM mass erase

maz3max, 2019-11-22:
I understand that the relevant code for **unlocking** NRF52 chips got copied from https://devzone.nordicsemi.com/f/nordic-q-a/12484/approtect-and-dap/47301 by @markrages.
When I took a look at it, I wondered why the addresses `MDM_POWER_EN` and `MDM_SELECT_AP` are not multiples of four. I am sure that part of the code currently writes stuff to invalid addresses and therefore does nothing.

This line:
https://github.com/blacksphere/blackmagic/blob/8a07f444351298f93b271ad6240ea9b3f8da4fe0/src/target/nrf51.c#L415
should be:
```cpp
adiv5_dp_write(ap->dp, ADIV5_DP_CTRLSTAT, (ADIV5_DP_CTRLSTAT_CDBGPWRUPACK | ADIV5_DP_CTRLSTAT_CDBGPWRUPREQ));
```

And that line:
https://github.com/blacksphere/blackmagic/blob/8a07f444351298f93b271ad6240ea9b3f8da4fe0/src/target/nrf51.c#L417
should be:
```cpp
adiv5_dp_write(dp, ADIV5_DP_SELECT, ADIV5_DP_BANK0);
```

UweBonnes, 2019-11-22:
This reads like you did not test the code and crosscheck with Nordic documents. I leave this Issue open for comments from other parties.

maz3max, 2019-11-22:
Taking your snarky comment as an invitation to delve deeper,
here are the relevant documents:
Product Spec.: https://infocenter.nordicsemi.com/pdf/nRF52832_PS_v1.4.pdf
Prod. Prog. Manual: https://infocenter.nordicsemi.com/pdf/nwp_027.pdf

Looking at the Prod. Prog. Manual, we should use a 15s timeout and reset the target after the procedure. Also, the registers are actually referred to by different names.
Testing on an NRF52832, I makes **no difference** if the aforementioned lines are commented out or not.
Also, using the CTRL-AP becomes flaky when the device turns into sleep mode.

esden, 2021-12-15:
Considering that this issue is bit stale. Is it still an issue @maz3max? Just want to make sure we still need to address this. :)

maz3max, 2021-12-16:
Hey @esden , I am not interested in this anymore. This could have been a contribution if addressed constructively when it was fresh.

closed 2021-12-16 by maz3max

esden, 2021-12-16:
@maz3max ok no problem. I am only now getting back into the swing of things. Please excuse Uwe's tone. This is not the attitude we want to project with this project. If you decide to get back to this we will welcome your contribution. Take care. :)