blackmagic-debug/blackmagic #563: A proper way to run periodic task in BMP firmware

katyo, 2019-12-01:
I trying to implement Segger's Real-Time Transfer protocol (or j-link RTT) in BMP firmware directly.
Because it requires asynchronous polling of target device memory I would like to find a proper way to run periodic tasks.

UweBonnes, 2019-12-01:
Did you have a look at the appropriate gdb_if_getchar() for your platform.

sidprice, 2019-12-01:
I had the same issue for my ctxLink probe (based upon BMP). My current solution is not as clean as I would like so I will be moving to a custom gdb_if.c module as @UweBonnes mentioned above.

katyo, 2019-12-02:
Because my RTT protocol implementation uses only `target_mem_read()`/`target_mem_write()` functions I would like to implement polling of buffers in platform independent way.

UweBonnes, 2019-12-02:
It will be very hard  to write a platform independent  code for polling working on BMP  microcontroller-hosted and pc-hosted platforms. As libopencm3 does not know threads to my knowledge, you would need a timer triggered interrupt routine or something similar on uc-host platforms. On pc-hosted platforms you would use an additional thread.

Otherwise, in the present code, both uc- and pc- hosted platforms idle in the gdb-if code.

katyo, 2019-12-02:
@UweBonnes,
Great thanks for good explanation.

closed 2019-12-11 by UweBonnes