blackmagic-debug/blackmagic #564: Just a Question

leogomide, 2019-12-02:
Greetings, 

Im analysing the bmpm_v2_1c_schematic file, and i see the "PWR_FLAG" icon at the schematic.

Im new to electronics, can somebody tell me why is the VCC pin tied to GND by this label?

Also many thanks for this amazing project :)

[image]

sidprice, 2019-12-02:
This is a convention with Kicad to indicate a power connection. It is often used when a number of connections of the schematic are defined as "inputs" or "passive" and it may appear to the error checker that the net has no source driving it.

leogomide, 2019-12-02:
So its just a indicator of a power line, they are not connected to each other, right?

sidprice, 2019-12-02:
That is correct, they indicate to the error checker that the net has power, but they are not connected to each other.

UweBonnes, 2019-12-02:
Unrelated to BMP

closed 2019-12-02 by UweBonnes

leogomide, 2019-12-02:
Thanks for the explanation @sidprice :)