blackmagic-debug/blackmagic #565: For looking swd target need send jtag_scan

boris92git, 2019-12-03:
Hi, I have the problem when I uses BMP with MCU ATSAMR34. 
Target connected with helping SWD interface, but so that to see its need send in gdb:
mon jtag_scan
And after:
mon swdp_scan
And then we see the target. If we skip the step jtag_scan, we don't see the target.

UweBonnes, 2019-12-03:
The ATSAMR34 is not  supported, Feel free to contribute support.

grevaillot, 2019-12-03:
@boris92git Supporting R34 should only be a matter of adding a couple of ids - It's basically a L21 with a radio.

boris92git, 2019-12-04:
@grevaillot 
You didn't understand me. 
I asked about another.
I need that see the target do  
1) mon jtag_scan
and before 
2) mon swdp_scan
if I miss the first step, the second command doesn't work.

UweBonnes, 2019-12-04:
JTAG is not often used an so may have problems. Feel free to tackle these problems.

esden, 2021-12-16:
Hi @boris92git unfortunately I don't have the hardware to test with so we can't address the issue ourselves. Can you take a look at the differences between the handling for jtag vs swd? Is it possible that the chip in question has some issues with it's debug state machine?

Usually the debug interface is in jtag mode by default and has to be switched over to swd. The swd scan does it. Is it possible that the chip has an errata about that problem? I recommend that you record the communication between bmp and your target to narrow down the issue. If you can figure out what exactly is going on we might be able to fix it.

dragonmux, 2024-11-09:
Please let us know if this is still a problem on the latest `main` which should have addressed the issues seen here. If we don't hear back in couple of weeks, we'll close the issue as done but if you then get back to this and it turns out it is still broken, we can re-open the issue.

dragonmux, 2024-11-20:
Closing this as done, but the note about letting us know if it's not actually fixed still applies.

closed 2024-11-20 by dragonmux