blackmagic-debug/blackmagic #570: [RFC] OpenOCD HLA interface driver for Blackmagic

walmis, 2019-12-17:
I've created a basic implementation for Blackmagic firmware itself and implemented a basic OpenOCD HLA interface driver for blackmagic probe.
On the probe itself, I've extended the remote.c interface with a new class REMOTE_HL_PACKET. 
It's currently in experimental stage, but seems to pass functional tests: eg single stepping, flashing, RTOS threads support, openocd built in memory transfer tests. 
Speed and performance feel similar to that of st-link. Transfer rate is about 30 KB/s. Half of that of direct gdb connection, but for RTOS support of OpenOCD I think it's a reasonable tradeoff.

You can find the code on my github:
https://github.com/walmis/openocd-blackmagic
https://github.com/walmis/blackmagic

UweBonnes, 2019-12-17:
Can you try
https://github.com/UweBonnes/blackmagic/tree/cmsis_dap
against yout firmware?

walmis, 2019-12-18:
Hi, I'm not really sure what to try on that branch as I don't have a CMSIS-DAP adapter.

UweBonnes, 2019-12-18:
There are( too?) many debugger variants. With your first post, I thought that you implemented a firmware for (Stlink)BMP to behave as a  OpenOCD HLA  adapter. I also thought that " OpenOCD HLA" is the same as a CMSIS-DAP adapter. I have to have a look at the code to understand more.

- What is the connection to Mubes PC-Hosted platform?
- Does it compile with mingw?
- Does it work on windows?
- Do you know the PC-STLINKV2 platform?
- Di you have a look at the command line interface for PC-Hosted BMP platforms?

walmis, 2019-12-18:
No, the BMP firmware core functionality is as-is, nothing changed. On the BMP side, the only addition are the remote api calls for target read mem/write mem/read reg(s)/write reg(s). This is the core functionality required for OpenOCD HLA(High level adapter layer). The magic happens inside OpenOCD as it has access to target registers and memory over BMP. BMP in this case works like a bridge.
To clarify: 
- my fork of BMP runs on stm32f103c8t6
- openOCD communicates with BMP over /dev/ttyACM serial
- GDB communicates over OpenOCD

hiviah, 2019-12-19:
This looks interesting. BMP has extra features, but I sometimes need the OpenOCD commands.

Note that when cloning recursively with submodules, you are missing autoreconf in `openocd-blackmagic/src/jtag/drivers/libjaylink` (thus no `configure` script and exit)

You could add this to README for running it for clarity (STM32F4x sample chip):

`openocd-blackmagic/bin/openocd -f share/openocd/scripts/interface/blackmagic.cfg -f share/openocd/scripts/target/stm32f4x.cfg`

However flashing doesn't seem to work (maybe it wasn't part of this code):

`/home/ondro/local/openocd-blackmagic/bin/openocd -f /home/ondro/local/openocd-blackmagic/share/openocd/scripts/interface/blackmagic.cfg -f /home/ondro/local/openocd-blackmagic/share/openocd/scripts/target/stm32f4x.cfg -c "init; reset halt; flash write_image erase build/firmware/firmware.bin.p1 0x08040000; flash write_image erase build/firmware/firmware.bin.p2 0x08120000; exit"`

`/home/ondro/local/openocd-blackmagic/bin/openocd -f /home/ondro/local/openocd-blackmagic/share/openocd/scripts/interface/blackmagic.cfg -f /home/ondro/local/openocd-blackmagic/share/openocd/scripts/target/stm32f4x.cfg -c "init; reset; exit"`

UweBonnes, 2020-03-26:
@hiviah: For target devices supported, your openocd commands  now collapse to
"blackmagic_stlinkv2 -a  0x08040000 build/firmware/firmware.bin.p1"
"blackmagic_stlinkv2 -a  0x08120000 build/firmware/firmware.bin.p2"
"blackmagic_stlinkv2 -R"

UweBonnes, 2020-04-12:
Your patch is usefull, even without using BMP with OpenOCD, as it will greatly improve command line programming speed when used like "blackmagic_hosted -d /dev/ttyACMx ><file.bin>. However it defeats another use of pc-hosted: Develop device algorithms without having to reflash the BMP. Can you think about adding a command line option to disable the high level access and in that case do only low level access?

UweBonnes, 2020-04-20:
Are you still listening?

 REMOTE_INIT_SWDP hard codes target one an provides a duplicate way to swdptap_init.
I think remotePacketProcessHL/REMOTE_INIT_SWDP should be removed and replaces by a pure call to  target_attach_n

UweBonnes, 2020-05-09:
https://github.com/UweBonnes/blackmagic/tree/multi implements remote high level interface in a way it is usable for BMP  pc-hosted. Comments welcome.

brainstorm, 2020-05-10:
@UweBonnes Adding an example session on https://github.com/UweBonnes/blackmagic/commit/d0cc2df00753d0806a8ad529c3826d94f905c971 (with both server/client involved) would help users getting started, IMHO.

UweBonnes, 2020-06-08:
@brainstorm:

Please have a look at platforms/hosted/Readme.md and tell me what is missing.

brainstorm, 2020-06-08:
@UweBonnes Just an example **`shell` session** with any hardware of your choice, as a guiding **example**.

UweBonnes, 2020-06-08:
For running with gdb, you need two shells: 
- one where you start the gdb server:
> blackmagic
Using 0483:374b 0670FF535651727067091832 STMicroelectronics STM32 STLink
Listening on TCP: 2000

- And the second where you start gdb:
> arm-none-eabi-gdb
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) tar ext :2000
Remote debugging using :2000
(gdb) mon s
Target voltage: 3.22V
Available Targets:
No. Att Driver
 1      STM32F74x M7
(gdb) 

The gdb windows is the same as with running firmware BMP but you must connect to :2000 instead of /dev/ttyACM0

Is this the example you need?

brainstorm, 2020-06-09:
It's not about what **I** need ;) I was just reviewing the docs from a technical/UX perspective for (other) people that might approach this project without having had contact with it before.

Yes, that looks better, I would perhaps remove all the GDB license banner in the interest of brevity but the interaction and outputs can help as a guide. Cheers!

UweBonnes, 2020-06-09:
Can upu please proof-read https://github.com/UweBonnes/blackmagic/tree/hosted? Thanks

brainstorm, 2020-06-12:
Sure! This paragraph is a bit too dense?:

> This builds the same GDB server, that is running on the Black Magic Probe. While connection to the Black Magic Probe GDB server is via serial line, connection to the PC-Hosted GDB server is via TCP port 2000 for the first GDB server and higher for more invokations. Use "tar(get) ext(ented) :2000" to connect. PC-hosted BMP GDB server can talk to the Black Magic Probe itself, ST-LinkV2 and V3, CMSIS-DAP, JLINK and FTDI MPSSE based debuggers.

The last part I would break it down into a list, typo on "invocations". I would remove the `target` explanation bit since it's already shown in the example.

The rest LGTM 👍

esden, 2021-12-16:
@UweBonnes Just to clarify. Is this branch referring to your OpenOCD support branch? What is the status of that project? What are the todo items that need to be done here?

ankostis, 2022-06-03:
Any hope of pushing your changes to BOTH upstream projects?

ikcalB, 2024-03-05:
hi guys,

is there any progress on this? using the BMP as a transport in openOCD could really push the spread of BMP. (allowing to support more different targets - as is my current use case)

dragonmux, 2024-03-05:
This is not something which is in the project roadmap to go fix/implement as we just don't have the people power to go after OpenOCD's contribution process while also trying to evolve BMD itself.

If it's a feature you want to see happen and you feel you have the skills and time to write an OpenOCD backend for BMD (using BMDA's remote protocol implementation as a template) then that would be very welcome by us - can't speak for OpenOCD there though. As noted in the post above yours, BMD itself is fully capable already, there's no code changes required our side to make something like that happen.