blackmagic-debug/blackmagic #571: #define SAMD_DID_SERIES_MASK		0x1F

hsn93, 2019-12-19:
im reading [samd21 datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/SAMD21-Family-DataSheet-DS40001882D.pdf) 

i think
#define SAMD_DID_SERIES_MASK		0x1F

should be 0x3F

UweBonnes, 2019-12-19:
Do you see any impact when changing the mask? Are there any parts that define 0x1f as mask?

hsn93, 2019-12-19:
i didnt check the impact.
i dont know about other parts but:
i know SAMD21, SAMD20 ,SAMD10, and SAMD11 are all series mask 0x3F

however, checking on datasheet SAMD11 seems like only two values are being used only

|  SERIES[5:0]  	|   Description                                            |   	     
|:-:	                |:-	                                                         |
|   0x0              |  Cortex-M0+ processor, basic feature set |   	
|  0x1               |  Cortex-M0+ processor, USB                    |   	
|  0x2-0x3F      |   Reserved                                                 |   	

 but this data isnt available on samd21 datasheet

esden, 2021-12-16:
Is there a good reason to proactively expand the mask from 0x1F to 0x3F? Should we leave it as it is until it turns out to be an issue with some future part or change it now? I am pretty much ok with both solutions considering that most of the bits seem to be reserved and not used on the currently existing parts.

dragonmux, 2022-06-04:
Looking this over, for correctness with the datasheet, we should use the 0x3F mask to future proof the code better. We'll get a PR in that closes this issue for v1.9

PR #1162 (2022-08-07): Fix: SAMD series mask
This PR fixes the SAMD series mask and does some rudimentary reformatting of the constants defined top of file.

This fixes #571.
closed 2022-08-07 by esden