blackmagic-debug/blackmagic #580: Cannot attach BMP to VM

Enzeys, 2020-01-11:
I developed my own custom PCB to use as a BMP and I can't seem to be able to connect it to my Debian VM - when I try to mount it I get the error 'E_INVALIDARG,' and tells me that the device is currently busy with a previous request.
It however works perfectly well with the windows version of GDB.

I tried using a USB Mass Storage Device example and it worked perfectly on both the host machine and on the VM without any errors - so I'm assuming this issue can be fixed by modifying the firmware to allow for this hardware configuration.

As an additional note I have a secondary BMP running on the BluePill development board and it works perfectly using the STLINK platform firmware.

On my custom BMP I'm using the STM32F103CBU6 as the MCU and the USB pins are attached directly to the corresponding pins on the micro-usb connector, with no other components in-between or in parallel on the data lines.

I'm using the firmware for the STLINK platform - slightly modified for custom LEDs, but this doesn't seem to be the source of it as the original firmware has the same issue.
As additional information: I upload it to my PCB via a secondary BMP - first uploading the blackmagic_dfu.elf (using objcopy to convert it from the bin file and 'changing' the .main section address to 0x8˙000˙000), then I upload blackmagic.bin to 0x8002000 through dfu-util.

UweBonnes, 2020-01-11:
Did you try a simple blinky program on that platform?
To flash DFU, use blackmagic_dfu elf file. Does DFU enumerate after a replug?

Enzeys, 2020-01-12:
I have tried a blinky program, blinking both of the LEDs, it worked perfectly. Also, as I said my custom BMP seems to work with the Windows version of GDB - currently I only have the BluePill handy also with BMP (stlink platform) installed on it, and when I try to probe the BluePill with my custom platform I get an error saying it received an "invalid ack." - does the BMP firmware use the SWD pins of the BluePill?
Both the DFU and full version of BMP cannot be enumerated on the VM, but works well on Windows.
As an unrelated question: is it possible to flash BMP firmware through an SWD connection only once or is the DFU way of flashing it required?

UweBonnes, 2020-01-12:
So you work with a linux VM under windows? And windows does recognize the BMP? Does youe VM monitor offere you the option to foreward USB devices? Does it offer the BMP? Did you give BMP USB controll to the VM? Does the Linux kernel recognize the BMP? What access rights does the resulting device have? Has your VM user access rights to that device?

Enzeys, 2020-01-12:
Yes, I work with a Linux VM under Windows. I was trying to get screenshots of the error message, but it seems now that the issue is gone. I now have access to the BMP on the VM and I can communicate with it through GDB.

I'm not sure what the cause of the issue could've been, as when I had the error I kept re-plugging the BMP, restarted my PC and so on, but the issue persisted. The issue is now gone after leaving the PC off and the BMP unplugged throughout the night.

When attaching the BMP through 'target extended-remote' however I sometimes get a peculiar error: "Device or resource busy," and I can only attach to it after issuing the command multiple times. When attached and trying to issue the swdp_scan command when connected to a BluePill I still get the exception: "SWDP invalid ACK."
I'm assuming this is a hardware issue - but I'm really not sure how as I kept all the signal traces short; This is however a separate issue so I guess I need to open a new thread and this one can be closed?

UweBonnes, 2020-01-12:
Often in Linux a "Modem manager "  server is running. After plugging in the USB device ( and after handing over the device from the host to the VM), the modem manager is some times busy with  the new uart device. Only after the modem manager has not found any user for the modem device, the device is ready for the user.
"SWDP invalid ACK." is normally the sign to start working with a scope and logic analyzer if things are as you expect.
Modem manager is needed for WifI/Bluetooth sticks, If you do not need that functionality, stop the modem manager service

Enzeys, 2020-01-12:
Thank you for the explanation for the modem manager, very useful info!
I'll check the signals with an oscilloscope soon then comment my findings and close the issue if the problem is from signal integrity.
I also noticed in the stlink firmware that there's something about setting PA15 to reconnect to the USB, my pcb doesn't have this functionality so could this cause any issues?

Enzeys, 2020-01-12:
Okay, something very peculiar is going on and I would appreciate any help.
Attaching the oscilloscope with its probe at 10X to SWDIO fixed the problem; I attached to the BluePill, uploaded and debugged a program without any issues. At 1X the ACK problem persists.

Despite not having any issues with flashing and debugging the waveform looks very horrible: there's a slight ~4MHz oscillation overlaying the signal (I'm using an 8MHz crystal so that might have something to do with it?), but there seems to be some sort of slew-rate issue when running the 'att 1' command. I attached two pictures showing both 'interferences:'
[image]
[image]

What could be the cause of this issue?
I'm assuming the 4 MHz interference is from the SWDCLK? But what could be the cause for the horrible SWDIO slew-rate? And why does it work with the oscilloscope probe at 10X? How can I fix this issue so that I won't need the oscilloscope attached whenever I want to use my BMP?

UweBonnes, 2020-01-12:
Bad or insufficient wiring or long cables?  Bad or insufficent decoupling?  Show your schematics and setup!

When going off board, we don't want fast signals, as that will ring. The waveform above looks good w.r.t the levels.

Enzeys, 2020-01-12:
The cables off-board are about 10cm long - I've used BMP with longer cables before so that can't be the issue.
Here are the schematics; 100nF capacitors (C1, 2, 3, 4) are attached as close as possible to the MCU's power pins:
[image]

And an approximate picture: yellow square is the location of the crystal on the other side of the board, the orange squares are the resistors for SWDIO and SWCLK.
[image]

I'm assuming that if I just remove the two resistors for SWD it should work correctly or could that cause any sort of issue when probing different targets?

Enzeys, 2020-01-12:
Upon removing the resistors and replacing them with solder bridges I can debug the BluePill without any issues with my custom BMP platform.
I'm still however worried about the following waveform interferences/issues:
[image]

I mean, if it works I guess it can't be that bad; but it isn't ideal.

UweBonnes, 2020-01-12:
The second red rectangle comes when either debugger and debugge release SWDIO and the SWDIO pull-up pulls the line high.

UweBonnes, 2020-01-12:
Rethink  the Reset circuit. A capacitor to VCC is contraproductive, It should go to ground. Look at the STM hardware reference manuals and examples.

closed 2020-01-12 by UweBonnes

Enzeys, 2020-01-13:
That's true; Could that be the cause for another issue I seem to be having?
That is, the blackmagicprobe's LED (stlink firmware) start's blinking at a constant rate and GDB reports that it cannot communicate with the target (BMP) anymore. I can fix this issue quite easily by simply unplugging and plugging it back in again, or detaching it from the VM and then back again.

UweBonnes, 2020-01-13:
Is the blinking the DFU bootloader blinking?Your reset circuit with a capacitor to VCC makes the Reset line with Power Up immediate release reset. Power on Reset will perhaps do the right thing, but try with a capacitor to ground and w/o additional pullup as in the Nucleo/Disco Boards etc.

Enzeys, 2020-01-14:
I'll have to do that on Friday, I hope that's the cause of the issue.
The LED that blinks is the 'led_idle_run' in platform.c.