blackmagic-debug/blackmagic #584: STM32L0: Cannot flash the chip anymore

almusil, 2020-01-22:
Hi,
I was able to flash the chip few times, then it suddenly "locked" itself and every attempt ends up like the snippet:
```
$arm-none-eabi-gdb -x Node/.gdbinit Node/node/node.elf
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from Node/node/node.elf...
Enabling target power
Assert SRST during connect: enabled
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      ARM Cortex-M M0+
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0x08000c42 in reset_handler () at ../../cm3/vector.c:68
68	../../cm3/vector.c: No such file or directory.
Loading section .text, size 0x57b8 lma 0x8000000
Loading section .data, size 0x1dc lma 0x80057b8
Start address 0x8000c38, load size 22932
Transfer rate: 131 KB/sec, 917 bytes/write.
Section .text, range 0x8000000 -- 0x80057b8: MIS-MATCHED!
Section .data, range 0x80057b8 -- 0x8005994: MIS-MATCHED!
warning: One or more sections of the target image does not match
the loaded file

Kill the program being debugged? (y or n) [answered Y; input not from terminal]
[Inferior 1 (Remote target) killed]
```

Any idea what might be wrong?

Thank you

UweBonnes, 2020-01-22:
You should tell what the program you flashed is supposed to to. I guess you put the CPU to sleep. During sleep,  the MCUID can not be read, so the chip can not be identified. Did you try connect under reset?

almusil, 2020-01-22:
Yes the program sleeps a lot. But I did set `mon connect_srst enable` is that right? Without this I was not able to even attach. I mean once I attach `load` says everything is fine but `compare-sections` throws mismatch, and unfortunately the program is still the same as before nothing has changed. 
Thanks

UweBonnes, 2020-01-22:
Please try https://github.com/UweBonnes/blackmagic/tree/demcr and report back

almusil, 2020-01-22:
Won't compile without applying:
```
diff --git a/src/target/cortexm.c b/src/target/cortexm.c
index f7d653a..4289f67 100644
--- a/src/target/cortexm.c
+++ b/src/target/cortexm.c
@@ -243,6 +243,7 @@ static void cortexm_priv_free(void *priv)
 
 static bool cortexm_forced_halt(target *t)
 {
+    struct cortexm_priv *priv = t->priv;
        target_halt_request(t);
        /* Request halt on reset */
        target_mem_write32(t, CORTEXM_DEMCR, priv->demcr);
```

But unfortunately even then it is the same story.

almusil, 2020-01-22:
On the other hand the PR #507 helped.

ijager, 2020-03-06:
I had the exact same problem as @almusil. 

My first workaround was to add the following to my `bmp.gdb` script:

```
mon tpwr disable
mon tpwr enable
``` 

But of course that doesn't work for battery-powered devices. So then I tested PR #507 and it also works for me!

UweBonnes, 2020-03-09:
Reworked #507 misses handling WFI and the needed setup of of RCC to write DEBUG_MCU. However the detection works so far that probe is called, at least what I tested. In that case, probe for L0/G0 can set up RCC and DEBUG_MCU. Will try the next days.

UweBonnes, 2020-03-09:
misses handling WFI for STM32L0/G0

esden, 2021-12-16:
Is this issue now resolved? Can we close it?

dragonmux, 2024-11-09:
Please let us know if this is still a problem on the latest `main` which should have addressed the issues seen here. If we don't hear back in a couple of weeks, we'll close the issue as done but if you then get back to this and it turns out it is still broken, we can re-open the issue.

dragonmux, 2024-11-20:
Closing as done, but the note about letting us know if it's not actually fixed still applies.

closed 2024-11-20 by dragonmux