blackmagic-debug/blackmagic #588: Not working on Blue Pill

a-downing, 2020-02-02:
The official releases here http://builds.blacksphere.co.nz/blackmagic/blackmagic_dfu-swlink.bin, and here http://builds.blacksphere.co.nz/blackmagic/blackmagic-swlink.bin aren't working correctly on two  STM32F103C8T6 Blue Pills that I've tried. Flashing the DFU image works with both openocd and st-flash. Connecting to USB shows:
```
[83109.756664] usb 3-1.5: new full-speed USB device number 125 using ehci-pci
[83109.856397] usb 3-1.5: New USB device found, idVendor=1d50, idProduct=6017, bcdDevice= 1.00
[83109.856400] usb 3-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[83109.856403] usb 3-1.5: Product: Black Magic (Upgrade), SWLINK, (Firmware 7a595ea)
[83109.856405] usb 3-1.5: Manufacturer: Black Sphere Technologies
[83109.856406] usb 3-1.5: SerialNumber: 2A76986C
```
Neither blackmagic_upgrade, or dfu-util work however. Both hang indefinitely with no errors. Flashing blackmagic-swlink.bin with openocd, or st-link both give flash verification errors. That part makes sense to me since the firmware is larger than 64k. I'm not sure if this just relies on there possibly being more flash than is documented in some cases?

The Blue Pills have 10k resistors on R10. I tried changing one of them to a 1.5k resistor, and it made no difference.

I've compiled from modified source where I removed some targets and added "-ffunction-sections -fdata-sections -Wl,--gc-sections" to src/Makefile to reduce the image size and was able to successfully flash blackmagic.bin with both openocd and st-flash. However, running blackmagic_upgrade compiled with the smaller firmware still hangs indefinitely. When connected to USB, dmesg shows:
```
[82985.648202] usb 3-1.5: new full-speed USB device number 123 using ehci-pci
[82985.747424] usb 3-1.5: config index 0 descriptor too short (expected 191, got 9)
[82985.747428] usb 3-1.5: config 1 has 0 interfaces, different from the descriptor's value: 6
[82985.748952] usb 3-1.5: New USB device found, idVendor=1d50, idProduct=6018, bcdDevice= 1.00
[82985.748955] usb 3-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[82985.748957] usb 3-1.5: Product: Black Magic Probe (SWLINK), (Firmware v1.6.1-409-g7a595ea-dirty)
[82985.748959] usb 3-1.5: Manufacturer: Black Sphere Technologies
[82985.748961] usb 3-1.5: SerialNumber: 2A76986C
```

No ttyACM devices are created.

JShabtai, 2020-02-03:
I'm not super familiar with this project, but from what I read (https://forum.pine64.org/showthread.php?tid=8816&pid=57095#pid57095) there are versions of the blue pill with 128k flash. So if your chip has only 64k, then you're out of luck.

I am hoping to be able to make a stripped down version that works on the blue pill, but have no idea if this is practical or not (would also depend on what you actually want to use it for, my aim is to program the Pine Time, which is an nrf52832)

UweBonnes, 2020-02-03:
Are you sure you have a genuine STM32F103x8? As explained many times before, both STM32F103x[8|B] are the same die. Only difference is that upper flash seem untested on the 64 k device. No active disabling has been noticed so long.
A GD32F103x8 has only 64 k, so it will not work. Other clones may behave more different.

If you through out algos, do not through out the pc-hosted parts. This will allow you to still programm families you have thrown out with pc-hosted for firmware BMP platform.

a-downing, 2020-02-03:
It does say STM32F103C8T6 on the IC package, but I am not sure if it's genuine. I think there is something else wrong here too besides that mine are only 64k. Even when I only flash blackmagic_dfu.bin, it doesn't work. It is recognized by both blackmagic_upgrade, and dfu-util, but both hang forever at 0% progress. It appears that blackmagic_upgrade is hanging on this line in upgrade/main.c `assert(stm32_mem_erase(handle, iface, LOAD_ADDRESS + offset) == 0);`. I am able to erase the flash with both openocd and st-flash.

I will have a Black Pill with STM32F103CBT6 to test this on soon. I'll modify the code to deal with the different pinning.

a-downing, 2020-02-03:
> I am hoping to be able to make a stripped down version that works on the blue pill, but have no idea if this is practical or not (would also depend on what you actually want to use it for, my aim is to program the Pine Time, which is an nrf52832)

You can definitely make it fit if you comment out the targets you don't need in src/target/cortexm.c around line 361, and add `-ffunction-sections -fdata-sections -Wl,-gc-sections` to CFLAGS in src/Makefile. That's how I was able to make blackmagic fit on my 64k device. I don't think this would break anything, but it could very well be why mine isn't working. I'm curious if it will work for you.

a-downing, 2020-02-03:
I just tried it on the Black Pill with 128k and everything worked perfectly. Is there anything in any of the code that depends on the flash actually being 128k, or does it all just need to fit? I still can't see why even the dfu bootloader didn't work on my 64k devices since it fits easily.

suibin0127, 2020-03-05:
I have some attempts to convert my blue pill board (STM32F103C8T6) as below.
1.  since my board USART doesn't work not able to boot by ST bootloader for the flash, i use J-link V9 to do the job with SWD connector. First, I program "blackmagic_dfu-swlink.bin" @0x08000000. 2nd, I program "blackmagic_swlink.bin" @0x08002000, have to change target from "103C8T6" to "103CB" which support 128K Flash for this large bin. Both steps works fine. But after that, when I change BOOT1 to 1 to force the DFU bootloader, i suppose to see 2 emulated USB virtual device in the device manager from WIN10, but nothing happened only the green led blinks.

Any step I missed or wrong ?  thanks.

UweBonnes, 2020-03-05:
Green LED blinking means that the bootloader is active. An that is what you requested with BOOT=1. Set BOOT=0 and restart and BMP should come up.

suibin0127, 2020-03-05:
> Green LED blinking means that the bootloader is active. An that is what you requested with BOOT=1. Set BOOT=0 and restart and BMP should come up.

thanks for comments, Yes, i think it is working, i also tried to set BOOT to 00, and the Green LED turn to be green no further blinking, then i tried a reset button, also no USB port detected.
I checked the win10 device manger, also switch to VM ware, use lsusb to check any device come up, but also don't show any device there...wierd... i think J-Flsah Tool is ok to flash as long as I select the right bin file (swlink dfu + swlink bin)....

suibin0127, 2020-03-05:
> thanks for comments, Yes, i think it is working, i also tried to set BOOT to 00, and the Green LED turn to be green no further blinking, then i tried a reset button, also no USB port detected.
> I [...]

One more thing is, my blue pill board USART not working, always get init device to fail when i try to boot from USART with different tool (stm32bootloader), i use a PL2303 USB /TTL converter for RS232 conneciton to PA9/PA10, so my quesiton is: does black magic firmware will use USART function to emulate the USB COM PORT ? if that is the case , might it will not able to use on my board. thanks.

UweBonnes, 2020-03-05:
Second last posting: Does BMP/swlink/bluepill bootloader come up with BOOT=1 and gets detected. Try to flash the binary when the bootloader is running with scripts/stm32mem.
Last posting: I do not know if you mean BMP/swlink/bluepill  UART or Bluepill/target uart.

suibin0127, 2020-03-05:
> Second last posting: Does BMP/swlink/bluepill bootloader come up with BOOT=1 and gets detected. Try to flash the binary when the bootloader is running with scripts/stm32mem.
> Last posting: I do not know if you mean BMP/swlink/bluepill UART or Bluepill/target uart.

for first suggestion, by understanding, i let the blue pill running under bootloader mode (BOOT1=1), the green led blink, then run python script to re-flash "swlink.bin", it show no DFU found.

for 2nd post, i mean the USART from Bluepill itself..

**[root@bogon scripts]# python3 stm32_mem.py  -m ../../blackmagic/blackmagic-swlink.bin 
-
USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
-
No DFU devices found!**

UweBonnes, 2020-03-05:
Does the device manager see the BMP/DFU device? If yes, configure your USB access, e.g. zadig

suibin0127, 2020-03-05:
> Does the device manager see the BMP/DFU device? If yes, configure your USB access, e.g. zadig
Device manage didn't see any DFU devices after scan and refresh , also tried Zadig, the list no any DFU device display, only have a camera.

BTW, do I need to install the dfu driver from "blackmagic.inf" & "blackmagic_upgrade.inf"?
i have  turned off win10  digital integrity check， looks still not able to install that driver,
will it be a problem for DFU issue ? Thanks.

suibin0127, 2020-03-05:
Device manage didn't see any DFU devices after scan and refresh , so from Zadig, the list no any DFU device display, only have a camera. Maybe no DFU driver for STM32, i also installed "DfuSe" driver, doesnt help ~~

suibin0127, 2020-03-05:
I found the problem, it is very basic mistakes i made. I use a micro-USB cable , but it was only have power lines, no data lines. I cut it and see only 2 wires there. So, after I switch to another good data cable it was able to recognize BMP!!

Thanks for your support, appreciated ! @UweBonnes  :)

USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
-
Device ID:       1d50:6017
Manufacturer:    b'Black Sphere Technologies'
Product:         b'Black Magic (Upgrade), SWLINK, (Firmware 7a595ea)'
Serial:          b'E4D295B7'
Invoking Application Device

UweBonnes, 2020-05-05:
Operator forgot to close.

closed 2020-05-05 by UweBonnes