blackmagic-debug/blackmagic #590: EFR32 support

denigmus, 2020-02-12:
Any plan to support EFR32 for at least batch programming? I see EFM32 is already supported.

UweBonnes, 2020-02-12:
BMP is a community effort. Device support is added via Pull Requests from owner of the hardware who implement and test the support.

kanflo, 2020-02-17:
Just out of curiosity, what does "batch programming" refer to? General EFR32 support is on master, flashing included.

denigmus, 2020-02-17:
Batch programming described here 
https://github.com/blacksphere/blackmagic/wiki/GDB-Automation

UweBonnes, 2020-02-17:
You can also use PC-hosted BMP platforms like "blackmagic_stlinkv2 bla.bin" to flash at 0x08000000 or with otional modifiers. But also gdb batch programming should work.

kanflo, 2020-02-17:
Does "regular programming" work for your device (ie. nor in batch mode) or is the chip completely unsupported (ie. not found when probing)?

denigmus, 2020-02-17:
I don't have this debugger at all and know nothing about it. :) I just 
review options for a batch programming of EFR32.

kanflo, 2020-02-20:
It will most likely work fine unless the firmware quickly enters EM4 as the BMP currently lacks support for the "AAP Expansion Sequence". I say "most likely" as the EFM/EFR:s I have encountered so far work fine but as always, YMMV.

UweBonnes, 2020-04-24:
Just stumble on this:
> grep -i -r efr32 target/efm32.c 
 * EFM32, EZR32 and EFR32 devices are all currently supported through
 * * EFR32BG13P532F512GM32 (EFR Blue Gecko)
        {16, "EFR32MG1P", 2048, 0x400e0000, true, 2048, 10240, "Mighty Gecko"},
        {17, "EFR32MG1B", 2048, 0x400e0000, true, 2048, 10240, "Mighty Gecko"},
        {18, "EFR32MG1V", 2048, 0x400e0000, true, 2048, 10240, "Mighty Gecko"},
        {19, "EFR32BG1P", 2048, 0x400e0000, true, 2048, 10240, "Blue Gecko"},
        {20, "EFR32BG1B", 2048, 0x400e0000, true, 2048, 10240, "Blue Gecko"},
        {21, "EFR32BG1V", 2048, 0x400e0000, true, 2048, 10240, "Blue Gecko"},
        {25, "EFR32FG1P", 2048, 0x400e0000, true, 2048, 10240, "Flex Gecko"},
        {26, "EFR32FG1B", 2048, 0x400e0000, true, 2048, 10240, "Flex Gecko"},
        {27, "EFR32FG1V", 2048, 0x400e0000, true, 2048, 10240, "Flex Gecko"},
        {28, "EFR32MG12P", 2048, 0x400e0000, true, 2048, 32768, "Mighty Gecko"},
        {29, "EFR32MG12B", 2048, 0x400e0000, true, 2048, 32768, "Mighty Gecko"},
        {30, "EFR32MG12V", 2048, 0x400e0000, true, 2048, 32768, "Mighty Gecko"},
        {31, "EFR32BG12P", 2048, 0x400e0000, true, 2048, 32768, "Blue Gecko"},
        {32, "EFR32BG12B", 2048, 0x400e0000, true, 2048, 32768, "Blue Gecko"},
        {33, "EFR32BG12V", 2048, 0x400e0000, true, 2048, 32768, "Blue Gecko"},
        {37, "EFR32FG12P", 2048, 0x400e0000, true, 2048, 32768, "Flex Gecko"},
        {38, "EFR32FG12B", 2048, 0x400e0000, true, 2048, 32768, "Flex Gecko"},
        {39, "EFR32FG12V", 2048, 0x400e0000, true, 2048, 32768, "Flex Gecko"},
        {40, "EFR32MG13P", 2048, 0x400e0000, true, 2048, 16384, "Mighty Gecko"},
        {41, "EFR32MG13B", 2048, 0x400e0000, true, 2048, 16384, "Mighty Gecko"},
        {42, "EFR32MG13V", 2048, 0x400e0000, true, 2048, 16384, "Mighty Gecko"},
        {43, "EFR32BG13P", 2048, 0x400e0000, true, 2048, 16384, "Blue Gecko"},
        {44, "EFR32BG13B", 2048, 0x400e0000, true, 2048, 16384, "Blue Gecko"},
        {45, "EFR32BG13V", 2048, 0x400e0000, true, 2048, 16384, "Blue Gecko"},
        {49, "EFR32FG13P", 2048, 0x400e0000, true, 2048, 16384, "Flex Gecko"},
        {50, "EFR32FG13B", 2048, 0x400e0000, true, 2048, 16384, "Flex Gecko"},
        {51, "EFR32FG13V", 2048, 0x400e0000, true, 2048, 16384, "Flex Gecko"},
        {52, "EFR32MG14P", 2048, 0x400e0000, true, 2048, 16384, "Mighty Gecko"},
        {53, "EFR32MG14B", 2048, 0x400e0000, true, 2048, 16384, "Mighty Gecko"},
        {54, "EFR32MG14V", 2048, 0x400e0000, true, 2048, 16384, "Mighty Gecko"},
        {55, "EFR32BG14P", 2048, 0x400e0000, true, 2048, 16384, "Blue Gecko"},
        {56, "EFR32BG14B", 2048, 0x400e0000, true, 2048, 16384, "Blue Gecko"},
        {57, "EFR32BG14V", 2048, 0x400e0000, true, 2048, 16384, "Blue Gecko"},
        {61, "EFR32FG14P", 2048, 0x400e0000, true, 2048, 16384, "Flex Gecko"},
        {62, "EFR32FG14B", 2048, 0x400e0000, true, 2048, 16384, "Flex Gecko"},
        {63, "EFR32FG14V", 2048, 0x400e0000, true, 2048, 16384, "Flex Gecko"},

UweBonnes, 2020-05-05:
Closing as no feedback . EFR should be supported and batch programming can be done with PC-hosted variants

closed 2020-05-05 by UweBonnes