blackmagic-debug/blackmagic #593: Compilation fail on macos

akamensky, 2020-03-04:
```
$ make PROBE_HOST=pc-hosted clean && make PROBE_HOST=pc-hosted
  CLEAN
 GIT include/version.h
  CC      target/adiv5.c
  CC      target/adiv5_jtagdp.c
  CC      target/adiv5_swdp.c
  CC      command.c
  CC      target/cortexa.c
  CC      target/cortexm.c
  CC      crc32.c
  CC      target/efm32.c
  CC      exception.c
  CC      platforms/pc/gdb_if.c
  CC      gdb_main.c
  CC      gdb_hostio.c
  CC      gdb_packet.c
  CC      hex_utils.c
  CC      target/jtag_devs.c
  CC      target/lmi.c
  CC      target/lpc_common.c
  CC      target/lpc11xx.c
  CC      target/lpc17xx.c
  CC      target/lpc15xx.c
  CC      target/lpc43xx.c
  CC      target/kinetis.c
  CC      main.c
  CC      morse.c
  CC      target/msp432.c
  CC      target/nrf51.c
  CC      target/nxpke04.c
  CC      platforms/pc-hosted/platform.c
  CC      target/sam3x.c
  CC      target/sam4l.c
  CC      target/samd.c
  CC      target/samx5x.c
  CC      target/stm32f1.c
  CC      target/stm32f4.c
  CC      target/stm32h7.c
  CC      target/stm32l0.c
  CC      target/stm32l4.c
  CC      target/target.c
  CC      platforms/pc/cl_utils.c
platforms/pc/cl_utils.c:255:37: error: missing field 'printf' initializer [-Werror,-Wmissing-field-initializers]
        struct target_controller tc = {NULL};
                                           ^
1 error generated.
make[1]: *** [cl_utils.o] Error 1
make: *** [all] Error 2
```

UweBonnes, 2020-03-04:
Can you try #594 and report back?

akamensky, 2020-03-04:
👍 that compiles and seems to work (to an extent of it working as described in the chat).

closed 2020-03-04 by UweBonnes