blackmagic-debug/blackmagic #597: Problems with blackmagic_dfu-stlink on robotdyn blackpill

ARKbr, 2020-03-05:
Greetings, 

I have flashed the blackmagic_dfu-stlink.bin using an original ST-LinkV2 and STMCubeProgrammer at address 0x08000000 of the black pill, and its a sucess

Then i flashed the blackmagic-stlink.bin at the address 0x08002000 and used the option byte command -d 1d50:6018,:6017 , and its a sucess again

[image]

But when i try to upload using arduino IDE, i get this errors:

```
`C:\Users\Usuario\AppData\Local\Arduino15\packages\STM32\tools\arm-none-eabi-gcc\8.2.1-1.7\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
COM21: No such file or directory.
"monitor" command not supported by this target.
Don't know how to attach.  Try "help target".
You can't do that when your target is 'exec'
warning: One or more sections of the target image does not match
the loaded file
Section .isr_vector, range 0x8000000 -- 0x800010c: matched.
Section .text, range 0x800010c -- 0x80055d8: matched.
Section .rodata, range 0x80055d8 -- 0x8005aa4: matched.
Section .init_array, range 0x8005aa4 -- 0x8005abc: matched.
Section .fini_array, range 0x8005abc -- 0x8005ac0: matched.
Section .data, range 0x8005ac0 -- 0x8005ccc: MIS-MATCHED!

The program is not being run.`
```

The BMP apears at COM21 and COM22, tried both and its the same:

[image]

Heres my arduino options:
[image]

The hardware scheme im using is this one:
[image]

I have used zadig to install libUsbK as the drivers and nothing changes

I have no experience in linux, so i dont have followed the CLI tutorials out there

Can someone guide me to solving this? 

Thanks in advance :)

UweBonnes, 2020-03-05:
Your connections are wrong! Bluepill exposed SWD/SWCLK on the four pin connector. Recheck the docs

UweBonnes, 2020-03-05:
B.t.w, you can use the SWLINK with a recent ST firmware as a BMP probe. Compile for PROBE_HOST=pc-stlinkv2 and run the resulting blackmagic_stlinkv2 executable

ARKbr, 2020-03-05:
i tried to reflash the binaries and tried the blackmagic-swlink.bin too and the following connections:

(just for exemple, the gnd and 5v pins where connected)
[image]

none of them worked :/

any other suggestion?

UweBonnes, 2020-03-05:
Where is the ground connection?

ARKbr, 2020-03-05:
> Where is the ground connection?

Its just an exemple, the 5v and gnd where connected

the user "mubes" helped me at the discord chat, i come to the following using blackmagic-stlink.bin and connecting with the below schematic:

[image]

i can connect to the probe:
```
(gdb) target extended-remote \.\COM21
Remote debugging using \.\COM21
```

i found the target device with the swo scan:
```
(gdb) monitor swdp_scan
Target voltage: unknown
Available Targets:
No. Att Driver
 1      STM32F1 medium density M3/M4
```

but i cannot attach to it:
```
(gdb) attach 1
Attaching to Remote target
Could not read registers; remote failure reply 'EFF'
(gdb)
```

i think it can be someting while flashing the firmware with st-link maybe, i have done this:
- full erase blackpill chip with stlink
- flashed blackmagic_dfu-stlink.bin at 0x08000000 with chip erase:
[image]

- flashed blackmagic-stlink.bin at 0x08002000 without chip erase:
[image]

is this ok?

ARKbr, 2020-03-06:
Finally found the sollution,

As i see in thi link [https://gitter.im/blacksphere/blackmagic/archives/2016/11/21](url), i have used the command  `set debug remote 1` and get this output:

```
(gdb) attach 1
Attaching to Remote target
Sending packet: $vAttach;1#37...Ack
Packet received: T05
Packet vAttach (attach) is supported
Sending packet: $qC#b4...Ack
Packet received:
Sending packet: $Hg0#df...Ack
Packet received:
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture>  <feature name="org.gnu.gdb.arm.m-profile">    <reg name="r0" bitsize="32"/>    <reg name="r1" bitsize="32"/>    <reg name="r2" bitsize="32"/>    <reg name="r3" bitsize="32"/>    <reg name="r4" bitsize="32"/>    <reg name="r5" bitsize="32"/>    <reg name="r6" bitsize="32"/>    <reg name="r7" bitsize="32"/>    <reg name="r8" bitsize="32"/>    <reg name="r9" bitsize="32"/>    <reg name="r10" bitsize="32"/[508 bytes omitted]
Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11...putpkt: Junk:
Ack
Packet received: mme="faultmask" bitsize="8" save-restore="no"/>    <reg name="control" bitsize="8" save-restore="no"/>  </feature></target>
Sending packet: $qXfer:features:read:target.xml:475,3fb#b6...Ack
Packet received: l
Sending packet: $g#67...Ack
Packet received: EFF
Could not read registers; remote failure reply 'EFF'
```
so, mubes and UweBonnes helped me, saying it was a **bad GND connection, faulty wires or something like this** and BINGO, thats the solution :D 

arduino ide and platformio still dont work with upload comand, but platformio works fine in debug mode

closed 2020-03-06 by ARKbr