blackmagic-debug/blackmagic #624: swlink: strange behavior on macos

skotopes, 2020-04-01:
Basically you can perform only one debug session. After you disconnect, bmp doesn't talk to gdb anymore.

Looks like there is a some kind of condition race with handling control requests on cdcacm side that leaves bmp in wrong state. So debugging show that bmp is trying to send data over closed serial port with cdcacm_gdb_dtr=0.

esden, 2021-12-16:
@skotopes Any progress on debugging this? Is it still an issue with the most recent firmware?

skotopes, 2021-12-16:
@esden no progress last I've checked.

I've abandoned ideas on fixing it, instead we've ported BM to esp-s2 and solved issue there(https://github.com/flipperdevices/blackmagic-esp32-s2). We use totally different stack and FreeRTOS, so no back port possible.

esden, 2021-12-16:
Ok, thank you! :) I am closing the issue then.

closed 2021-12-16 by esden