blackmagic-debug/blackmagic #625: cortexm: Atmel SAMD21G18A detected as nRF51

WallaceIT, 2020-04-02:
Atmel SAMD21G18A gets detected as an nRF51 after a SWDP scan:

`monitor swdp_scan`

Expected result:

```
Available Targets:
No. Att Driver
1      Atmel SAMD21G18A (rev D) M0+
```

Obtained result:

```
Available Targets:
No. Att Driver
 1      Nordic nRF51 M0+
```

Probably a side-effect of commit a336ac20847b6f2059a8f77f774362699cfbd32b.

A simple patch could be the following, but I do not have any nRF51 hardware for testing it - if someone can confirm it is working, I'll open a merge request:

```patch
---
 src/target/cortexm.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/src/target/cortexm.c b/src/target/cortexm.c
index b81b4d1..7619361 100644
--- a/src/target/cortexm.c
+++ b/src/target/cortexm.c
@@ -369,8 +369,8 @@ bool cortexm_probe(ADIv5_AP_t *ap, bool forced)
 	PROBE(lpc43xx_probe);
 	PROBE(sam3x_probe);
 	PROBE(sam4l_probe);
-	PROBE(nrf51_probe);
 	PROBE(samd_probe);
+	PROBE(nrf51_probe);
 	PROBE(samx5x_probe);
 	PROBE(lmi_probe);
 	PROBE(kinetis_probe);
-- 
2.25.1
```

Thank you

UweBonnes, 2020-04-02:
Moving around probing is not sustainable.

Can you compile and run a PC hosted BMP platform, e.g. "make PROBE_HOST=pc-hosted" and "./blackmagic_hosted -t <-d COMx>" where COMx is the gdb com on windows, On linux with only one BMP connected the device is not needed and auto-deteced. On STM 32 e.g, I get
> ./blackmagic_hosted -t
...
ROM: Table BASE=0xe00ff000 SYSMEM=0x1, PIDR 0x00000a0411
...
where 0xa0 gives 0x20 for STM and 0x411 is a STM32F407. #507 now can be much more selective  in probe:

        if (ap->ap_designer == AP_DESIGNER_STM) {
                PROBE(stm32f1_probe);
                PROBE(stm32f4_probe);
                PROBE(stm32h7_probe);
                PROBE(stm32l0_probe);   /* STM32L0xx & STM32L1xx */
                PROBE(stm32l4_probe);
        } else {

Probably all ATMEL parts can also have their own clause. But that change is intrusive. So I need help from owners of such board before I can roll out.

WallaceIT, 2020-04-03:
Here it is the output of the hosted BMP:

```
AP   0: IDR=04770031 CFG=00000000 BASE=41003003 CSW=03000040
ROM: Table BASE=0x41003000 SYSMEM=0x1, PIDR 0x000009fcd0
ROM: Table BASE=0xe00ff000 SYSMEM=0x1, PIDR 0x04000bb4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008) -> cortexm_probe
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a)
 2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b)
 ROM: Table END
1 0x41006000: Debug component - CoreSight MTB-M0+ (Simple Execution Trace) (PIDR = 0x04001bb932)
ROM: Table END
```

I think the relevant line is the following one:

```
ROM: Table BASE=0x41003000 SYSMEM=0x1, PIDR 0x000009fcd0
```

with 0x1F being the Atmel device ID and 0xCD0 the part number.

I also added the dump to #507

UweBonnes, 2020-12-09:
Shoud be fixed with #507 applied

closed 2020-12-09 by UweBonnes