blackmagic-debug/blackmagic #629: semihosting endianness

koendv, 2020-04-05:
Hi.
Thanks for writing this debugger probe.

I'm using a Raspberry Pi 4 as host, running  arm-none-eabi-gdb 8.3.0, a Black Magic Probe 1.6.1 (from git) running on a STM32F103 Blue Pill as debugger, and a STM32F103 Blue Pill as target. 
I'm using semihosting to debug a "blinking led" program. [f1-bluepill-blink.zip](https://github.com/blacksphere/blackmagic/files/4433463/f1-bluepill-blink.zip)
Character-at-a-time output works, string output doesn't.
Comparing qemu and BMP output:

Using qemu: 
```
$ xpack-qemu-arm-2.8.0-9/bin/qemu-system-gnuarmeclipse  \
> --verbose --verbose \
> --board BluePill \
> -d unimp,guest_errors \
> --image f1-bluepill-blink.elf \
> --semihosting-config enable=on,target=native \
> --semihosting-cmdline test 6

main(argc=2, argv=["test", "6"]);
Hello ARM World!
Standard output message.
Standard error message.
System clock: 72000000 Hz
[led:red off]
[led:red on]
[led:red off]
Second 1
[led:red on]
[led:red off]
Second 2
[led:red on]
[led:red off]
Second 3
[led:red on]
[led:red off]
Second 4
[led:red on]
[led:red off]
Second 5
[led:red on]
[led:red off]
Second 6

QEMU semihosting exit(0)
```

Using BMP:
```
$ arm-none-eabi-gdb 
(gdb) target extended-remote /dev/ttyBmpGdb
(gdb) monitor swdp_scan
(gdb) attach 1
(gdb) set debug arm on
(gdb) file /home/koen/Downloads/f1-bluepill-blink.elf
(gdb) load
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/koen/Downloads/f1-bluepill-blink.elf 
Standard output message.
Standard error message.
```
In both cases - qemu and bmp - the led blinks, so the program runs. It's the debugging output that's missing.

If I run bmp with "(gdb) monitor debug_bmp enable", I get the following output:
```
koen@raspberrypi:~ $ cat < /dev/ttyBmpTarg 
*** Unsupported packet: qC
*** Unsupported packet: qOffsets
*** Unsupported packet: Hg0
syscall 0lx15 (20000088 4f 1 0)
syscall 0lx1 (80023d0 0 3 1)
syscall 0lx1 (80023d0 4 3 1)
syscall 0lx1 (80023d0 8 3 1)
syscall 0lx4 (6e69616d 67726128 2c313d63 67726120)
syscall 0lx4 (6e002222 67726128 2c313d63 67726120)
syscall 0lx4 (a3b295d 67726100 2c313d63 67726120)
syscall 0lx4 (6c6c6548 5241206f 6f57204d 21646c72)
syscall 0lx4 (a 53464320 203d2052 38302520)
syscall 0lxc (2 0 3 0)
syscall 0lx9 (2 0 3 0)
syscall 0lx5 (2 20000408 19 1)
m packet: addr = 20000408, len = 19
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx5 (3 20000373 1 2)
m packet: addr = 20000373, len = 1
syscall 0lx4 (74737953 63206d65 6b636f6c 3237203a)
syscall 0lx4 (6f636553 3120646e 6b63000a 3237203a)
syscall 0lx4 (6f636553 3220646e 6b63000a 3237203a)
syscall 0lx4 (6f636553 3320646e 6b63000a 3237203a)
syscall 0lx4 (6f636553 3420646e 6b63000a 3237203a)
syscall 0lx4 (6f636553 3520646e 6b63000a 3237203a)
syscall 0lx18 (0 0 0 0)
```
Let's take the first write 
```
syscall 0lx4 (6e69616d 67726128 2c313d63 67726120)
```
and translate the argument from hex to ascii:
```
koen@raspberrypi:~ $ xxd -r
0: 6e69616d 67726128 2c313d63 67726120
^D
niamgra(,1=cgra
```
This is ```main(argc=1, arg``` but in the wrong endianness.
What am I doing wrong?

koen

UweBonnes, 2020-04-05:
Start with the low byte of a uint32_t  and work up to the highest byte. That way, things look right.

koendv, 2020-04-05:
The same binary produces the correct semihosting output when run through qemu.

UweBonnes, 2020-04-05:
I do not see binary output in the qemu log. What I see i obvious interpreted output, probably observing little endianness.

koendv, 2020-04-15:
You're tight. There's no output because BMP does not implement the write0 syshosting call.

UweBonnes, 2020-04-15:
What is needed to implement?

koendv, 2020-04-15:
This patch implements SYS_WRITE0, SYS_EXIT, and SYS_GET_CMDLINE:
[bmp_patch.txt](https://github.com/blacksphere/blackmagic/files/4482393/bmp_patch.txt)

After this patch, the test program runs:
`
koen@raspberrypi:~/src/sh/blackmagic $ gdb -q
(gdb) target extended-remote /dev/ttyBmpGdb
(gdb) monitor swd
(gdb) attach 1
(gdb) file f1-bluepill-blink.elf 
(gdb) load
(gdb) set args x 4
(gdb) run
main(argc=2, argv=["x", "4"]);
Hello ARM World!
Standard output message.
Standard error message.
System clock: 72000000 Hz
Second 1
Second 2
Second 3
Second 4
_exit(0)

Program received signal SIGTRAP, Trace/breakpoint trap.
report_exception (reason=131110) at ../system/include/arm/semihosting.h:137
137	in ../system/include/arm/semihosting.h
(gdb)

`

UweBonnes, 2020-04-15:
I needed two compilation changes. Please review
https://github.com/blacksphere/blackmagic/pull/639
and comment if this should go into git head. It takes about 700 Bytes. Is the dummy heap command needed?

koendv, 2020-04-15:
Checking everything it seems I made an error. This is the correct patch, with SYS_HEAPINFO removed. 
[bmp_patch.txt](https://github.com/blacksphere/blackmagic/files/4482552/bmp_patch.txt)
It's just that I'm working on getting a few more semihosting calls running.

UweBonnes, 2020-04-15:
Can you clone BMP and work on a branch so that I can easily pull?

koendv, 2020-04-15:
Like [this](https://github.com/koendv/blackmagic) ?

UweBonnes, 2020-04-16:
Please include #include <ctype.h> for isxdigit, as gcc fails on you definition.

gcc also fails in target.c:
target/target.c: In function 'target_set_cmdline':
target/target.c:410:2: error: 'strncpy' specified bound depends on the length of the source argument [-Werror=stringop-overflow=]
  410 |  strncpy(t->cmdline, cmdline, len);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
target/target.c:407:12: note: length computed here
  407 |  len_src = strlen(cmdline)+1;
      |            ^~~~~~~~~~~~~~~

Look at my branch how I corrected that. Otherwise you probably did not check " “Allow edits from maintainers.” to I can not push my changes to you branch. if you do a PR, please provide as one patch, as long as no other areas are involvd.

koendv, 2020-04-19:
Thank you, very helpful. I've done a PR, with "allow edits from maintainers".

Also, I've attached a zip of my testbench. It's an stm32duino sketch; consider it a "proof of concept".  
[Semihosting-190420.zip](https://github.com/blacksphere/blackmagic/files/4499686/Semihosting-190420.zip)

koendv, 2020-04-23:
Closed with #641

closed 2020-04-23 by koendv