blackmagic-debug/blackmagic #631: SAMD5X Support

vslinuxdotnet, 2020-04-05:
Hello ,

Do Black Magic supports SAMD5X ?
I try the run a simple blink program in PB22, but the program don't run, maybe I need help...
See the GDB terminal:
```
remote debugging using /dev/ttyACM0
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Microchip SAMD51J19A (rev A) M3/M4
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0xfffffffe in ?? ()
(gdb) monitor serial
Serial Number: 0x070282555339463532202020ff0c173a
(gdb) monitor erase_mass 
Erase successful!
Erase successful!
(gdb) file /home/user/samd51_test.ino.hex
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from /home/user/samd51_test.ino.hex.ino.hex...(no debugging symbols found)...done.
(gdb) load
Loading section .sec1, size 0x2a9c lma 0x4000
Start address 0x437c, load size 10908
Transfer rate: 42 KB/sec, 909 bytes/write.
```
I need to convert it to GDB, In j-link:
```
erase
w2 0x41004000, 0xA51A
loadbin bootloader-grandcentral_m4-v3.6.0.bin, 0x0
verifybin bootloader-grandcentral_m4-v3.6.0.bin, 0x0
r
```

Any help to debug/run something in this CPU ?
Thank you.

UweBonnes, 2020-04-05:
Yes, the chip is detected. Normally  do
> arm-none-eabi-gdb
 tar ext /dev/ttyACM0
 att 1
 load
 compare section
 r
I do not know what "w2 0x41004000, 0xA51A" does for your device.
If you compile a PC-Hosted BMP, you can also do
> blackmagic_xxx -a 0 bootloader-grandcentral_m4-v3.6.0.bin
to write and to verify
> blackmagic_xxx -a 0 -V bootloader-grandcentral_m4-v3.6.0.bin

For blackmagic_hosted however speed is much less than when running in gdb.

vslinuxdotnet, 2020-04-05:
After some flash tests with SAMD21/SAMD51, I figure out what is the problem:
```
run
Starting program: /home/user/samd51_test.ino.hex
Program received signal SIGSEGV, Segmentation fault.
```
The program dies at run LOL... the hex or elf code generated by Arduino to SMAD51 can't run...
I load the elf bootloader from Adafruit to this CPU (M4) and them type: run.
The console goes the wait mode :) I check in bootloader code what was the boot led, add it and the led fade in/out so it works! :)

That is my experience.
Thanks

UweBonnes, 2020-04-06:
In do not understand your last posting. Do you have problems or are your problems solved. But anyways, the problems seems to be with your understanding of the tools, but not related direct to the BMP. Learn more about debugging e.g. at https://github.com/blacksphere/blackmagic/wiki etc.

vslinuxdotnet, 2020-04-07:
Hello, 
I have problems but now are solved. I was expected more debug in GDB in the flash job.
Yes the problems ATM was not BMP related.
Thank You

UweBonnes, 2020-04-09:
No BMP problem visible

closed 2020-04-09 by UweBonnes

kiranshila, 2021-05-31:
Wait so does BMP work with SAMD5X boards? The wiki does not seem to indicate this.

UweBonnes, 2021-05-31:
Did you try with a recent firmware? Test first, ask later...

kiranshila, 2021-05-31:
I'm planning on buying one if it supports it.