blackmagic-debug/blackmagic #632: Unable to flash second second sector on H750

hydra, 2020-04-06:
ok, so bear with me here.

The STM32H750VBT6 is advertised as having a single 128KB flash sector, but in reality it has 1MB.

With my previous build of blackmagicprobe I was able to erase BANK 1 and flash and run code from at least the first 4 sectors, my test project is about 512KB.

With the current master of blackmagicprobe, using the same debugger only the first 128KB sector is flashed.

Here's a view of the memory after erasing bank 1 using `monitor erase_mass` and uploading code via gdb's `load` command.
Bad (846dadcc39df5783988c7f030f722ab55568e6e5):

[image]

Here's the same thing, but uploaded using my older binary.

Good:
[image]

Any ideas where to start fixing this?

UweBonnes, 2020-04-06:
Git Bisect is one possibility. "git log -p src/raget/stm32h7.c" and trying to understand the changes is another possibility.
commit 56fb0f77663f9db2898692eae7445c79a30cf345 (origin/stm32f730_h750)
Author: Uwe Bonnes <bon@elektron.ikp.physik.tu-darmstadt.de>
Date:   Thu Feb 21 18:49:25 2019 +0100

    Handle STM32F730 and STM32H750.
    
    Flash sector calculation was wrong with small flash sizes.

looks suspicious. Probably the naximum map should be kept for H73|50, but perhaps a warning printed...

hydra, 2020-04-06:
yeah, I'm unable to bisect easily as my working branch is very old and had other changes.  I'll take a look at that commit and go from there.

UweBonnes, 2020-04-06:
Can you try https://github.com/UweBonnes/blackmagic/tree/h750?

Otherwise, for bisect, clone in a new directory and work there. No danger for your working tree.
Otherwise, it is good to have own changes in own tracking branches and a clean master. When master changes, rebase your branches.

hydra, 2020-04-06:
heh, I just created and tested this, and then saw your message above.  I'll look at that now.

https://github.com/blacksphere/blackmagic/pull/633

hydra, 2020-04-06:
I'll update the PR a little and add a DEBUG message like in your branch.

hydra, 2020-04-06:
done, tested, works.

hydra, 2020-04-06:
Note: I've never tried accessing the second flash bank, so I only add bank 1 in my PR.

UweBonnes, 2020-04-06:
Can you writ something to the second bank? Or is the second bank really not accessible?

hydra, 2020-04-06:
I have not tried, don't currently have time to test that.  If I get time to test I'll open another PR with my findings.

UweBonnes, 2020-05-04:
Should be fixed with commit ada17ada23cfde53191c806b78271a767fb87100. Re-open if not

closed 2020-05-04 by UweBonnes