blackmagic-debug/blackmagic #635: It is nescessary to fine decode FLASH size?

UweBonnes, 2020-04-06:
Hello,

today, two PRs popped up that tackle with flash size for different chip variant in some chip family. We go into great deapth to decode these size. However most vendors produce only one chip and on the variants sold with less flash, a smaller flash size is announced and the flash is probably untested but mostly functional. 
By setting up the memory map according to the sizes announced by vendor, we inhibit eventual use of that flash, if the user decides to use this flash at his own risk.

I propose that we announce the highest available size for a variant and drop the fine decoding for the memory sizes.

Any opinions?

sidprice, 2020-04-06:
I think allowing access to the full flash is a good idea. Ideal would be to check the vendor size and if flashing fails let the user know this could be the reason. However, I don't think there is really a way to do this, other than through debug messages.

UweBonnes, 2020-04-21:
Done with #645. StTM32L4 still does flashsize decoding.

closed 2020-04-21 by UweBonnes