blackmagic-debug/blackmagic #636: CDCACM issues in WSL

Gussy, 2020-04-11:
Some here may already be following along with the issue that the BMP has with [WSL](https://docs.microsoft.com/en-us/windows/wsl/faq) with locking up GDB when freshly connected, background here: https://github.com/microsoft/WSL/issues/3099

Since I've found what appears to be a fix, I wanted to bring the issue up here to see if this is something that is both possible and welcomed to fix in the BMP source. Happy to submit a PR if it's something we have control over as it's lower friction to make a change here than in a WSL build.

UweBonnes, 2020-04-11:
There are known problems on MacOS. A PR is welcome!

Gussy, 2020-04-12:
The "fix" for WSL is to comment out this line which sets `cdcacm_gdb_dtr` to `0`: https://github.com/blacksphere/blackmagic/blob/master/src/platforms/common/cdcacm.c#L438

It looks like that line came from https://github.com/blacksphere/blackmagic/commit/adabaa759296b8f2470d7c562963d4e34dc35b19 which fixed "bug 3307433" 9 years ago. No idea what the bug was since that's not a GitHub issue.

It would be nice to get some kind of fix into `master` that doesn't break some other functionality. I'm also content maintaining a WSL patched binary for myself and my teams' probes.

@UweBonnes Is the MacOS problem you're referring to the `/dev/tty.*` vs `/dev/cu.*` issue? I'll check to see if this has any affect.

UweBonnes, 2020-04-13:
Any fix for one system seems to stall on other systems. So we need to really understand the problems. Other isses are: 

The MacOs issue is #608 
There is an old issue #41 

Can you try after fresg replug in gdb:
(gdb) set debug remote 1
(gdb) tar ext /dev/ttyACM0

On linux I see:
Remote debugging using /dev/ttyACM0
Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Ack
Packet received: PacketSize=400;qXfer:memory-map:read+;qXfer:features:read+
Packet qSupported (supported-packets) is supported
...

Do you perhaps have a GDB debug build? Can you look where the open hangs?

esden, 2021-12-16:
Should this be handled differently on each OS and setup?

The MacOS issue with the buffered vs unbuffered serial interfaces is intrinsic to how they work and can not be solved. All we can do is document it and hope that people read it.

In case of WSL I can imagine that they have some buffering in place too. It would be good to check. If we can't fix it on our end it is best to document the issue and hopefully let people know how to work around it.

dragonmux, 2022-08-01:
@Gussy there was recently work done (for v1.8) on the CDC ACM implementation in #1014 - is this still an issue after that, or can we close this issue as solved?

dragonmux, 2024-11-09:
Given this issue has sat open with nobody saying it's still an issue, we're going to close it as fixed in v1.10. If it turns out that this is still an issue, please let us know and we'll re-open the issue.

closed 2024-11-09 by dragonmux