blackmagic-debug/blackmagic #649: BlackMagic probe with AC6 Ide and other IDEs

YvesLafont, 2020-04-29:
Hello, i work with Blackmagic 2.1 and stm32F4 and stm32H7  boards.
I use AC6 IDE, but i AC6 send a gdb "target-connect remote" command, seeming obsolete, and not recognized by BMP.
I would know IDE list for stm32, where BMP is full functional (except STM32Cube ! )
Thank you very much for feedback, i know that is not a BMP problem, but, if someone can help ?
Greetings
YL

UweBonnes, 2020-04-29:
GDB does not have any "target-connect remote" command at all. So I think this is an issue with the IDE, not BMP. Please close the issue and perhaps try to get help on the Discord channel. See the gitter channel for an invitation.

closed 2020-04-29 by YvesLafont

JojoS62, 2020-10-12:
BMP can be used with Eclipse or VSCode based IDE.
E.g., in AC6/STM32CubeIDE you need to create a new launch configuration with template 'GDB Hardware Debugging'
Uncheck 'Use remote target' and enter path to gdb or arm-none-eabi-gdb if it is in the search path.

[image]

in startup settings, you have to enter your connection settings for com port and swd/jtag scan

[image]