blackmagic-debug/blackmagic #663: on-probe traceswo decoding

koendv, 2020-05-13:
This is a patch that does traceswo decoding in the bmp itself. The decoded stream is written to the bmp usb serial port. To see the swo stream, on linux type ``cat /dev/ttyBmpTarg``. On Windows, use PuTTY to connect to the usb serial port, I suppose.
This patch adds a ``decode`` option to the ``traceswo`` command, e.g.:
``monitor traceswo 2250000 decode`` decodes the full swo stream and prints it to the usb serial port. swo stream is 2250000, asynchronous. 
``monitor traceswo 2250000 decode 0 2`` will print only swo stimulus channels 0 and 2.
``monitor traceswo 2250000`` switches decoding off. The undecoded traceswo stream is written to the trace port.
If you omit the baudrate, the baudrate defaults to 2250000.
On platforms that support Manchester coding, no baudrate is needed, e.g. 
``monitor traceswo decode`` prints the swo stream on the serial port. swo stream is manchester coding.
You can abbreviate things, eg. ``mo tr d``
I use this on my stm32f103. I find it sufficient for the simple but common case where output is a single stimulus channel. I have not tested this on other platforms. I do not know if it works with Manchester coding. I'd like to hear from you what happens.
Attached: patch and binary for stlink (stm32f103). Note: disconnect traceswo stream before upgrading firmware. 
[blackmagic-traceswo-patch05.txt](https://github.com/blacksphere/blackmagic/files/4623275/blackmagic-traceswo-patch05.txt)
[bmp-stlink.zip](https://github.com/blacksphere/blackmagic/files/4623276/bmp-stlink.zip)

UweBonnes, 2020-05-13:
Please join the BMP discord channel.

From the old gitter channel: "You can find the Discord link here: https://1bitsquared.com/pages/chat".

As with semihosting, trace is not something I use often."Mubes"  is the Trace specialist and his orbuculum the reference trace data handler. I need his opinion before I consider merge.
How does data rate compare between undecoded and decoded stream? 
How does binary size compare with and w/o thi patch? 
And why as a patch and not as some branch ?

mubes, 2020-05-14:
I think this will be really useful for the simple (normal!) use case. Unfortunately it loses the usb serial port, but that's an acceptable compromise since we can't create additional serial ports on this platform (out of usb endpoints).  Good work, thanks for this.

Can you please submit it as a PR so we can review the code more easily?

koendv, 2020-05-14:
OK, done. https://github.com/blacksphere/blackmagic/pull/664#issue-417925446

UweBonnes, 2020-05-16:
Merged with #664

closed 2020-05-16 by UweBonnes