blackmagic-debug/blackmagic #666: Modify blackmagic to connect over Bluetooth instead of USB

HarjitSi, 2020-05-26:
The Blackmagic adapter uses USB to connect to the host. I'd like to replace this with a Bluetooth (BT) link.

So, on the PC, I would point gdb to <something> which would use the PC's internal Bluetooth device. On the target board would be a Bluetooth device on it (ex. nRF52832 such as Taiyo Yuden's EYSHSNZWZ) and it would connect to the target microcontroller I want to debug. The connection between the Bluetooth device and the target microcontroller could be SWD, JTAG, etc.

I want the link to be performant i.e. push the BT link to high throughput.

I would be very grateful for pointers on where to start.

Thank you.

UweBonnes, 2020-05-26:
For a wireless BMP, look at Sid Price CTXLink http://www.sidprice.com/ctxlink/. It is based on BMP, but with no code in master BMP repo. You can do something like that yourself, probably with a lot of effort needed.

markrages, 2020-05-29:
My wireless BMP: https://github.com/markrages/blackmagic/blob/esp8266/src/platforms/esp8266/Readme.md

HarjitSi, 2020-05-30:
Thank you for the pointers. 

For many debugger solutions, the transport used to talk to talk to the debugger is USB with HID or custom protocol over HID or serial port or TCP, etc.

I do not have the expertise to write a custom USB driver or serial driver. So, I'm thinking of doing the following:

1) Use the gdbserver from BMP or pyOCD (https://github.com/mbedmicro/pyOCD)
2) Have that talk to the host PC's Bluetooth and create a tunnel to a Bluetooth dongle
3) On the Bluetooth dongle run BMP target interface code or DAPLink target interface code

I'll have to create my own protocol between the host PC's Bluetooth or see if there is a serial interface protocol over BLE available.

Thoughts?

I would love to make something JLink compatible but I don't know enough about writing a USB

markrages, 2020-05-31:
You don't need any custom driver with any of the BMP options presented to you.

The original BMP shows as a standard serial port, using the CDC-ACM USB driver already built into your PC.

And the wireless options show as a TCP network port over WiFi.

gdb can connect to a serial port or a network port natively.  But this is not true for BLE, because there is no standard BLE serial port profile

So your plan is more difficult than the options already presented to you.  You will have to write a driver program to translate your custom BLE protocol to a network socket.

HarjitSi, 2020-05-31:
> You will have to write a driver program to translate your custom BLE protocol to a network socket.

I agree. I'm speculating but I think BMP code has to be written such that it has two heads - serial or network socket. These, then munge the commands and transport them to the probe's back end which then executes the commands and then takes responses from the targets and the process is reversed.

I would love pointers on which files to look at for the network socket code, the code that interfaces the serial / network socket code to the back end and then the back end SWD/JTAG/etc. code.

sidprice, 2020-05-31:
I have a product, ctxLink, that already operates over USB and Wi-Fi. The repository is here https://github.com/sidprice/blackmagic/tree/ctxLink_master

HarjitSi, 2020-05-31:
@sidprice ctxLink supports two topologies:

1: Host <- USB -> ctxLink (STM32F401) <- SWD | JTAG -> Target
2: Host <- WiFi -> ctxLink (ATWINC1500 <-> STM32F401) <- SWD | JTAG -> Target

From reading about BMP, it shows up as a GDB server on a serial port that you can connect to with GDB. NOTE: I'm ignoring the SWO serial port for now.

Since you didn't want to (or need to) have a host driver, the host discovers the ctxLink over WiFi and connects to it. Then, in GDB, you point it to the ctxLink wireless device. 

Then, on ctxLink, you had to add code to the STM32F401 to initialize and configure the ATWINC1500. The STM32F401 is involved in making ctxLink discoverable and for establishing the connection to the host.

GDB sends WiFi packets to the ATWINC1500 and those are transported over the SPI port between the ATWINC1500 and the STM32F401 and the STM32F401 interprets them and then, maybe munges them, and then shoves them into the equivalent of the serial stream for the USB version. And on traffic on the way back from the target, the reverse occurs?

Please note: I don't know if the traffic that goes over WiFi is exactly the same as if a serial link is connected.

Can you please fix up the above description and add/share high level details? Can you share what aspects of BMP software you had to modify? And which files? I'll then poke around there.

Thanks!

sidprice, 2020-05-31:
@HarjitSi I think, since you have access to the code you can figure this out for yourself. I do not have time to do your work for you!

markrages, 2020-05-31:
One way to think of black magic probe firmware is as a C library that translates one standard format (JTAG and SWD ARM debug commands) to another format (GDB extended remote protocol).

The JTAG and SWD commands can be accomplished with basic "set a bit, clear a bit" GPIO operations.  So they are not difficult to port to any microcontroller.

The GDB protocol is a serial protocol.  You can connect it to an actual serial port, or a TCP network socket, based on your system and requirements.  The protocol is described at https://www.embecosm.com/appnotes/ean4/embecosm-howto-rsp-server-ean4-issue-2.html and at in the official GDB documentation.

Because it is a C library, it can be compiled for about any architecture.  Here is an ESP01 module running BMP firmware, debugging a Bluepill board:

[image]

HarjitSi, 2020-06-01:
@markrages thank you for the detailed and kind response!

@sidprice thank you; I understand. I was hoping to learn if my mental model was correct.

All the best! Give me a few weeks/months to design, build, test and get this up and running!

closed 2020-06-01 by HarjitSi