blackmagic-debug/blackmagic #682: reset pin for stlink clone on PB6 not considered

v01d, 2020-06-23:
I just installed BMP from master to an stlink clone. This variant has reset line on PB6. Looking at the code for stlink platform it seems RST is not considered on this port as a possibility. I modified SRST definition to be on PB6 and I can succesfully reset the target.
This variant is green PCB, single LED on top and pinout appears to be:

(top)
5V
3V3
SWIM
GND
NRST
(bottom)
5V
3V3
CLK
GND
DIO

UweBonnes, 2020-06-24:
"Stlink clone" can mean anything! I have never seen an STLINKV1/2/2.1 with SRST on PB6. PB6 is SWIM RST.

v01d, 2020-06-24:
This is the typical aluminium case stlink v2 clone, sold as SWD/SWIM capable.
Should this be used to update the stlink pinout for this version or you mean it is handled elsewhere?

There's a schematic for this variation here: https://www.hobbiton.be/blog/repurpose-stlink/stlink2.svg
Reference with pictures: https://hackaday.io/project/162597-st-link-clone-repurposing

UweBonnes, 2020-06-24:
Does the original ST firmware work on that chip? Find bootloaders https://github.com/Krakenw/Stlink-Bootloaders.git.
However I think that board is a bad clone, changing pinout for no reason. I do not want a new platform for such a board. Perhaps if you extend the reset procedure to additional handle reset on PB6, such a patch has minimal impact and could be acceptable.

v01d, 2020-06-24:
Actually, the original firmware that it came with also did not use the reset line. I used official ST tool to upgrade firmware and now the line is used. 

I just modified the reset line definition on BMP and that's it. But I guess it should be detected somehow.

UweBonnes, 2020-06-24:
What ST firmware worked on that chip ? V2 or V2.1

v01d, 2020-06-24:
The ST tool gives: V2J37S7 STM8/STM32 Firmware. So it seems to recognize this variation.

UweBonnes, 2020-06-24:
That does not answer the question about the type recognized V2 or V2.1

esden, 2021-12-16:
Is this still an issue? Is there some actionable item here? Or can we close this issue?

dragonmux, 2023-09-09:
This has now been addressed in #1460, closing as fixed.

closed 2023-09-09 by dragonmux