blackmagic-debug/blackmagic #683: Unable to compile with PROBE_HOST=stlink on master

thatch, 2020-07-01:
Full log at https://gist.github.com/thatch/6ad9b6c71109b80fa87321e4f5b6a99f,

```
make PROBE_HOST=stlink
...
  LD      blackmagic.elf
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: timing_stm32.o:(.bss+0x0): multiple definition of `running_status'; platform.o:(.bss+0x8): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:92: blackmagic.elf] Error 1
make: *** [Makefile:25: all] Error 2
```

Looks like the relevant lines are in `platforms/stlink/platform.c` and `platforms/stm32/timing_stm32.c`; changing the reference in the latter to extern makes it compile (because `timing_stm32.c` doesn't include `timing_stm32.h`, I don't really understand what the convention is, this hack made it compile again).

I'm on Archlinux, using `arm-none-eabi-gcc 10.1.0-1`

UweBonnes, 2020-07-01:
Did you make clean before?

thatch, 2020-07-01:
Yes, I tried `make clean`, as well as recloning to make sure:

```
$ cd $(mktemp -d)
$ git clone https://github.com/blacksphere/blackmagic
$ cd blackmagic
$ git submodule init
$ git submodule update
$ make PROBE_HOST=stlink
```

UweBonnes, 2020-07-01:
You have a picky compiler! What compiler, what version?
Try https://github.com/UweBonnes/blackmagic.git

thatch, 2020-07-01:
Compiler is `arm-none-eabi-gcc 10.1.0-1` and compiling from https://github.com/UweBonnes/blackmagic.git works.

UweBonnes, 2020-07-02:
Fixed by  #684

closed 2020-07-02 by UweBonnes