blackmagic-debug/blackmagic #685: arm-cortexm3-eabi-gcc: Command not found

phillowcompiler, 2020-07-03:
I tried to build the firmware as
`make CROSS_COMPILE=arm-cortexm3-eabi-`

But cannot...

```
 GIT include/version.h
  CC      target/adiv5.c
make[1]: arm-cortexm3-eabi-gcc: Command not found
Makefile:95: recipe for target 'adiv5.o' failed
make[1]: *** [adiv5.o] Error 127
Makefile:23: recipe for target 'all' failed
make: *** [all] Error 2

```
Please tell me what is "arm-cortexm3-eabi-gcc" command...

my gcc-arm-none-eabi is 15:6.3.1+svn253039-1+10.

UweBonnes, 2020-07-03:
Where did you get that strange "CROSS_COMPILE=arm-cortexm3-eabi-" information from? I do not find in the repo or wiki.
If the "arm-none-eabi-" toolchain is available on your system, "make" is enough to build for BMP native.

phillowcompiler, 2020-07-03:
Thank you!

I follow this.
https://github.com/blacksphere/blackmagic/wiki/Hacking

Yes, only
`make`
is done!

Thank you

closed 2020-07-03 by phillowcompiler