blackmagic-debug/blackmagic #686: Hardware unreachable after dfu update

llllVIIII, 2020-07-03:
Hi it seems I updated the blackmagic probe with the wrong fw bin file via dfu_util. Now I am unable to update it any more. Is it possible to flash it via a Jlink?

UweBonnes, 2020-07-03:
Replug and press the "force bootloader entry" buttom. Then dfu transfer the right image. Report success or post output when problems happen.

llllVIIII, 2020-07-03:
So do I understand it correct that I have to flash it with offset of 0x8000000 and not 0x8002000?

UweBonnes, 2020-07-03:
"dfu-util -l" show the area reachable by dfu. BMP firmware loads to 0x08002000 or in the case you compiled for stlink  with "make PROBE_HOST=stlinl ST_BOOTLOADER=1" at 0x08004000 to allow to upload with stlink-tool to an stlinkv2 with original ST-stlinkv2 bootloader.

llllVIIII, 2020-07-03:
Okay, I have the original black magic Probe and compiled it via "make" I have the arm-none-eabi tool chain in my path.

llllVIIII, 2020-07-03:
So I successfully flashed the new FW on my BMP. I would suggest to update the Wiki:

https://github.com/blacksphere/blackmagic/wiki/Upgrading-Firmware

Here is says you should flash the blackmagic-native.bin. But the makefile does not generate suche a file but a plain blackmagic.bin file. Also more up to date information was found here:

https://github.com/blacksphere/blackmagic/wiki/Hacking

UweBonnes, 2020-07-03:
Please read Upgrading-Firmware again!

llllVIIII, 2020-07-03:
Well, you are absolutely right. I should have read it more carefully. Thanks anyway for the quick replay and your patience 😊

closed 2020-07-03 by llllVIIII