blackmagic-debug/blackmagic #687: Unable to connect to SAM3U2 via SWDP

Enzeys, 2020-07-06:
Recently I made a new PCB using the ATSAM3U2EA-CU, which is a 100 ball BGA package. The PCB only has a pinout for power, SWDP, and reset. I tried connecting to it through mon swdp_scan, but it didn't find anything, as if there was no device connected at all. The device appeared to be working correctly on the PCB, it only drew about 1mA of current on startup and the internal 1.80V regulator outputed 1.80V as it should (measured on a decoupling capacitor.)

The [**datasheet**](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-6430-32-bit-Cortex-M3-Microcontroller-SAM3U4-SAM3U2-SAM3U1_Datasheet.pdf) doesn't seem to mention anything weird going on with the SWDP pins, except that they expect some sort of magic packet to switch them from JTAG-mode to SWDP-mode, which I assume blackmagic is doing already(?).

I would appreciate any sort of help on how to further troubleshoot this issue. Thank you.

UweBonnes, 2020-07-06:
If nothing is detected, double check your setup.
If the device runs at 1.8 Volt, you need a probe that is able to levelshift, as the BMP native.  If you use the BMP native, does the power led light up?

Otherwise, "blackmagic/src/target> grep -i -I 3u2 -r" does not matvhj anything, so I guess nobody wrote BMP support for the ATSAM3U2EA up to now.

What BMP version do you use? Can you use PC-Hosted for easy debug output?

Enzeys, 2020-07-06:
@UweBonnes 
Some internal circuitry runs at 1.8V and all external pins (except for 1.8V LDO decoupling & output) are 3.3V, so that shouldn't be an issue.

The file "blackmagic/src/target/sam3x.c" seems to implement support for SAM3xxE, but doesn't appear to be sending out the sequence - unless that's done somewhere the JTAG/SWDP code assuming it's not only a quirk for this mcu to switch from jtag to swdp.

I'm using a custom-made PCB to host the BMP firmware; I built the firmware using PROBE_HOST=stlink, and the only thing I changed in the code is the pin for the status LED. The command 'mon version' reports Black Magic Probe (Firmware v1.6.1-404-gc3a3f77-dirty) (Hardware Version 0). I really don't want to go through the trouble of PC-Hosting it if possible.

So far I've only been able to test my BMP board on SWDP-only targets (stm32f103, efm32zg) and it worked well. The ATSAM3U2EA is the first target I've (tried) probing that has JTAG enabled by default. The datasheet states that it expects a sequence on the two SWDP pins to disable JTAG and enable SWDP communication. I'm assuming the firmware doesn't send out this sequence? If that's the case I'm hopeful it should be an easy patch.

UweBonnes, 2020-07-06:
Is JTAGSEL connected and connected in the right way? At what level  is VDDBU connected.

Enzeys, 2020-07-06:
JTAGSEL is left unconnected. The datasheet says it has a permanent internal 15 kOhm pulldown and that it can be left unconnected for "normal use."
VDDBU and all other power lines are connected to 3.3V, except for VDDCORE and VDDPLL which are connected to the internal LDO (which is set to 1.8V by default.)

UweBonnes, 2020-07-06:
Read again carefully:
"The JTAGSEL pin is used to select the JTAG boundary scan when asserted at a high level. It integrates a permanent pull-down resistor of about 15 k to GNDBU, so that it can be left unconnected for normal operations."

Enzeys, 2020-07-06:
Isn't JTAG boundary scan a different feature than the debugging port? If I understand correctly the JTAGSEL pin should be low in order to enable SWDP.
I believe this diagram from the same datasheet shows that.
[image]

UweBonnes, 2020-07-06:
Switch to SWD/JTAG is done in adiv5_adiv5_jtag/swpd.
Are you sure the JTAG pins are really connected to the BGA and there is no shortcircuit? Do you see some reaction from the SAM TDO with a scope?

Enzeys, 2020-07-07:
After reading this section of adiv5_swdp.c:
[image]
it seems to only send 50 clock cycles after sending out the sequence, but the datasheet specifies to send MORE than 50 cycles with TMS = 1:
[image]
Could this cause it not to switch to SWDP properly? I also noticed the bit order is reversed (sending out 0xE79E instead of 0x79E7), but I'm guessing this is due to SWDP specifications?

The board is currently not available to me at the moment, I'll get access to it in the evening, but I can't probe TDO as I left it unconnected.

UweBonnes, 2020-07-07:
All Cortex M and SAM chios others and me used so long switched with 50 TMS 1 cycles. But to make sure, change in the first or  third and in line 6 16/18 to a higher value up to 32
Test with PC-Hosted, so you do not need to reflash the BMP and you get easy debug output with -v 31
You should scope test TMS, best in relation to TCK.

Enzeys, 2020-07-08:
Thank you so much for suggesting testing the protection diodes! Turns out the package wasn't soldered on correctly and thankfully I had a spare board, which does get detected as "Atmel SAM3U M3/M4."

closed 2020-07-08 by Enzeys

Enzeys, 2020-07-09:
@UweBonnes on an unrelated note when running a program, the chip encounters a 'SIGTRAP' at 0x0018004C, which is located at the beginning of internal ROM. Would you happen to know what could be causing this?

Enzeys, 2020-07-10:
@UweBonnes On an unrelated note: when I flash the SAM3U2E with my code, no matter what it may be (even just 'b .' infinite loop as the first instruction), the code executes for a while, but then I recieve a SIGTRAP at a very large address such as 0xe008e008. Do you know by chance what could be causing this?

UweBonnes, 2020-07-12:
I am no debug guru. Perhaps https://stackoverflow.com/questions/3475262/what-causes-a-sigtrap-in-a-debug-session is a starting point.