blackmagic-debug/blackmagic #691: cl_opts.opt_mode is set to BMP_MODE_DEBUG using an assignment

HarjitSi, 2020-07-12:
In hosted\Platform.c, platform_init(), initializes cl_opts as follows:
 `	BMP_CL_OPTIONS_t cl_opts = {0};`

If enum bmp_cl_mode in Cl_utils.h was changed such that BMP_MODE_DEBUG wasn't the first item:
`enum bmp_cl_mode {
	BMP_MODE_DEBUG,'

This would be a fun bug to chase down.

Suggestion: explicitly initialize opt_mode to BMP_MODE_DEBUG either in Platform.c: platform_init() or Cl_utils.c: cl_init()

I'd submit a PR but I don't have the project building or the ability to test it. I'm still learning the code base.

UweBonnes, 2020-07-12:
Thanls for pointing out. I will try to keep in mind and fix when I touch those files again.

esden, 2021-12-16:
If this is a good suggestion we should implement that and make a patch, waiting until we "touch the code" will just leave this issue sitting around forever.

@HarjitSi did you make progress learning the codebase? Are you interested in taking a stab and writing a patch that implements this?

HarjitSi, 2021-12-18:
I have not had a chance to work in this area and probably will not until February or March. I need make the board, bring it up, etc. Sorry about that. If this is still open in that timeframe, we can revisit.

dragonmux, 2022-08-01:
This needs patching by direct assignment in https://github.com/blackmagic-debug/blackmagic/blob/main/src/platforms/hosted/cli.c#L237 which is where the cl_opt structure is actually initialised and configured now.

PR #1161 (2022-08-07): hosted/cli: Made sure the command line options structure is initialised with opt_mode = BMP_MODE_DEBUG before args parsing
This PR addresses the robustness point bought up in #691 so the BMDA CLI isn't left to the whims of the compiler or some future reorganisation that might change the command line mode enum

Fixes #691
closed 2022-08-07 by esden