blackmagic-debug/blackmagic #701: Cannot access global memory with stm32H743 target

bobbatcomcastdotnet, 2020-07-26:
I am having a problem with my BMP accessing global variables. My hardware is based on the STM32H743 chip. The application I am debugging is rather large (~300K code). I am using a native BMP with upgraded firmware:

Black Magic Probe (Firmware 54ee00b) (Hardware Version 3)
Copyright (C) 2015  Black Sphere Technologies Ltd.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

The same problem occurs with a Bluepill version of the BMP.

The error GDB returns is "access to memory location 0xxxxx is denied". However, if I do a memory dump I can access that memory address without problems.

This problem does not always occur and it only seems to happen when accessing global variables, particularly structures. I am using the GDB 'print /x variable` command to access the variables. In addition I have only seen this problem occur when debugging the STM32H743 code. I have never seen this problem when debugging my STM32F407 based code which is functionally the same as the STM32H743 version. Has anyone else experienced this problem using the BMP to debug an STM32H7 family chip?

UweBonnes, 2020-07-27:
Please run bmp-hosted with debugging enabled and try to find the offending section. Perhaps this can give us some hints.

bobbatcomcastdotnet, 2020-07-28:
> Please run bmp-hosted with debugging enabled and try to find the offending section. Perhaps this can give us some hints.

Is there a Windows version of the bmp-hosted version? If so are there instructions as to how to build/run it? I am currently quite busy attempting to build a HID composite host driver for the STM32H7 so I don't know when I'll be able to get this task done. As you may know, there is very little libopencm3 support for the STM32H7. I can continue to debug my code with the jlink probe.

UweBonnes, 2020-07-28:
When you have that problem via  'print /x variable` , resolve the address via "p &variable" and try to dump them memory via "x <address>"

Do me another favour: Compile bmp-hosted and rune bmp-hosted with the jlink. Flashing will be slow but normal debugging should work. Is the behaviour reproduced?

bobbatcomcastdotnet, 2020-07-28:
> When you have that problem via 'print /x variable` , resolve the address via "p &variable" and try to dump them memory via "x <address>"
> 
> Do me another favour: Compile bmp-hosted and run bmp-hosted with the jlink. Flashing will be slow but normal debugging should work. Is the behaviour reproduced?

Once again I must ask, Is there a Windows version of the bmp-hosted version? If so are there instructions as to how to build/run it? My environment is all Windows based.

UweBonnes, 2020-07-28:
Please give https://github.com/UweBonnes/blackmagic/tree/artefacts/src blackmagic.exe a try. Libuse and libftdi and zadig for the devices are needed.

sidprice, 2020-07-28:
@bobbatcomcastdotnet I have hosted running on Windows. To build from the current master:

cd src
make clean
make CC=gcc PROBE_HOST=hosted

That works for me.

Then start it "blackmagic -d YourComPort

[image]

Then you can connect with GDB ARM "target extended-remote :2000

bobbatcomcastdotnet, 2020-07-29:
I downloaded https://github.com/UweBonnes/blackmagic/tree/artefacts/src blackmagic.exe and associated dlls.
When I ran the app I get the following:

Microsoft Windows [Version 10.0.18363.959]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Users\Bob>cd\apps\bmp_hosted

C:\Apps\BMP_Hosted>blackmagic.exe -d COM1
Using 0000:0000
Could not open \\.\COM: 2

C:\Apps\BMP_Hosted>

I have my BMP connected and it is on COM1. Any suggestions?

UweBonnes, 2020-07-29:
I cut off the 1 from COM1 by error. Try again.

bobbatcomcastdotnet, 2020-07-29:
Do you mean download the exe again and then retry?

bobbatcomcastdotnet, 2020-07-29:
I downloaded the new ex and retries it. I get the following results:

C:\Apps\BMP_Hosted>blackmagic -d COM1
Using 0000:0000
Serial write failed 121, written 0
Timeout on read RESP

C:\Apps\BMP_Hosted>

UweBonnes, 2020-07-29:
Rub with -v 31 added to get a hint what is going wrong. And be sure that your BMP firmware is recent, so that remote is available.

bobbatcomcastdotnet, 2020-07-29:
If I use the following command line I get:

C:\Apps\BMP_Hosted>blackmagic.exe -d -v31 com1
Using 0000:0000
Could not open \\.\-v31: 2

My BMP firmware version is:

Black Magic Probe (Firmware 54ee00b) (Hardware Version 3)

sidprice, 2020-07-29:
It should be -> blackmagic.exe -d  com1 -v31

bobbatcomcastdotnet, 2020-07-29:
Using Sid's command line I get:

C:\Apps\BMP_Hosted>blackmagic.exe -d com1 -v31
Using 0000:0000
+#!GA#
Timeout on read RESP

Does this help?

UweBonnes, 2020-07-30:
This looks like no either no BMP is connected to COM1 or the firmware on the BMP is not recent and does not support the remote protocoll.

bobbatcomcastdotnet, 2020-07-30:
I downloaded the latest nightly build for the BMP and reloaded it onto the BMP using dfu-util. I confirmed it is connected to COM1 (i.e. I can debug using com1). The results I get when I run the blackmagic.exe app is the same as before:

C:\Apps\BMP_Hosted>blackmagic.exe -d com1 -v31
Using 0000:0000
+#!GA#
Timeout on read RESP

Any ideas...?

sidprice, 2020-07-30:
@UweBonnes Is there a reason you have Bob using the branch in your repo and not the master? I am running the master here on Windows 10.

UweBonnes, 2020-07-30:
The original request was to run with jlink and bmp-hosted to see if the access problem is also with BMP-Jlink-hosted. Bob also is on windows and did not offer to compile himself.

Please offer him your binary.

sidprice, 2020-07-30:
Well it seems that the master branch is now broken because I see the same issue as Bob. Let me regress and get a build that works. I don't have time to debug it right now.

sidprice, 2020-07-30:
@UweBonnes OK, here is the issue, I now remember mentioning this a while ago. In order to use blackmagic.exe you need to FIRST connect to the BMP using GDB, then exit GDB, and blackmagic.exe will connect.

@bobbatcomcastdotnet Launch arm-none-eabi-gdb.exe and connect to the BMP with "target extended-remote //./COM1". Exit GDB and try "blackmagic -d COM1"

bobbatcomcastdotnet, 2020-07-30:
@sidprice That certainly fixed the problem :) When I run the blackmagic app I now get the following response:

C:\Apps\BMP_Hosted>blackmagic -d COM1
Using 0000:0000
Remote is Black Magic Probe 71e5b4d
Listening on TCP: 2000

Now all I have to do is figure out how to use the new TCP access. As you may remember Sid I use my own IDE to do all of my STM32 development work. The BMP support I\my debugger currently has was not designed to use a TCP port so the IDE debug code does not currently have the means to handle this scenario. But it will as soon as I can get to it.

Thanks for your help Sid :)

sidprice, 2020-07-30:
@bobbatcomcastdotnet Happy you are connected, sorry my brain fade lost the vital data 😞

UweBonnes, 2020-07-30:
instead of tar ext com1 use tar ext :2000

bobbatcomcastdotnet, 2020-07-30:
OK, I've got everything setup to run BMP-Hosted.  Everything is apparently correct until I try to execute the GDB command 'monitor swdp_scan' at which point I get the following from blackmagic.exe:

C:\Apps\BMP_Hosted>blackmagic -d com1
Using 0000:0000
Remote is Black Magic Probe 71e5b4d
Listening on TCP: 2000
 0xe0041000: Debug component - Unknown (PIDR = 0x04001bb975 DEVTYPE = 0x13 ARCHID = 0x4a13)
Timeout on read RESP

C:\Apps\BMP_Hosted>

At which point blackmagic terminates. Any ideas...?

UweBonnes, 2020-07-31:
The abort is strange. Is that reproducable? Add debugging to see where it happens. Otherwise I will add the unknow component soon.

UweBonnes, 2020-07-31:
Unknown PIDR is fixed by #706

bobbatcomcastdotnet, 2020-07-31:
@UweBonnes  Okay, what is the next step for me?

UweBonnes, 2020-07-31:
As I told before: The abort is strange. Is that reproducable? Add debugging to see where it happens.

UweBonnes, 2020-07-31:
B.t.w; Are you sure you don not overclock the device, e.g. run an old version with 480 MHz or all those overdrive related bit are set the right way and wait states are enough?

bobbatcomcastdotnet, 2020-07-31:
@UweBonnes The abort happens every time I attempt to debug my application so yes it is reproduceable.  I am running the app at 480mHz. Now that you mention it I don't know is the STM32H743 chip is the correct version to support 480mHz. I am using clock code developed by Brian Viele for the libopencm3 project. Perhaps this is an area to investigate before we go any further with this process. However, this doesn't explain why the jlink probe appears to work? This is the exact reason I decided to try the jlink, to see if I had some kind of hardware issue.

Forgive me for my ignorance but you stated "Add debugging to see where it happens" in your comment. How do I do that? Is there a command line option that needs to be added?

UweBonnes, 2020-07-31:
Add "-v 31", and log the output to a file.

UweBonnes, 2020-07-31:
And look if you have  a Rev Y dev or a Rev V device.

bobbatcomcastdotnet, 2020-07-31:
I will try the -v31 option a little later today.

Is there a way to programmatically determine the device rev? I can't read the version # from the actual chip itself.

UweBonnes, 2020-07-31:
bit 8..15 in DBGMCU_IDCODE?

bobbatcomcastdotnet, 2020-07-31:
@UweBonnes Yeah, I found that just after I wrote the last message :( I added code to read the rev and it is 0x2003 so the chip is a revision V which means it is 480 mHz compatible.

bobbatcomcastdotnet, 2020-07-31:
@UweBonnes I ran the app again and produced the following output
[test.txt](https://github.com/blacksphere/blackmagic/files/5007147/test.txt)

UweBonnes, 2020-07-31:
Another thing to check: Is the SWD connection between your BMP and the target board stable? Enough ground lines, sensible long and no shacky connections? Does the target power the level shifters on the BMP with the right VDDIO? What is VDDIO by the way? And can you test with SWD from another Nucleo/Disco board? 

Otherwise, is your program open? I can test with a H743 disco and a H745 disco.

bobbatcomcastdotnet, 2020-07-31:
@UweBonnes The SWD connection between the BMP and the target is stable  (a 4" ribbon cable). It has three ground lines (in the ribbon cable). The target power for the level shifters on the BMP is supplied by the target (with the VREF signal) which is reported as 3.3V. I have used the same BMP (and ribbon cable) with several other boards (both F1 and F4). I have also used the BMP with a Nucleo-H743ZI2 board although on a different project. When I use the jlink probe I am using the same ribbon cable.

Unfortunately, the project is not open for distribution.

Did the test.txt file I sent have any useful information?

UweBonnes, 2020-07-31:
Where and how did you send?

bobbatcomcastdotnet, 2020-07-31:
I sent it here as an attachment to the message sent 2hr ago.

UweBonnes, 2020-07-31:
Did not look carefully enough.
The stall still happend during swd scan. I have no idea!

bobbatcomcastdotnet, 2020-08-01:
@UweBonnes Something I neglected to mention which may or may not have meaning. My IDE executes GDB using the mi2 interpreter. This might cause the blackmagic app trouble??? In fact, it might also be the reason I am having trouble accessing global variables (if the actual commands sent to the BMP are not handled properly).

UweBonnes, 2020-08-01:
BMP does not support MI2 to my knowledge

sidprice, 2020-08-01:
@bobbatcomcastdotnet MI2 is the protocol between a client application and GDB, BMP uses the Remote Serial Protocol, which is for GDB to GDBServer communications.

Does your IDE launch GDB and communicate with that, or is it connecting directly to BMP?

bobbatcomcastdotnet, 2020-08-01:
@UweBonnes  Why not? The interpreter used by GDB should have no effect on the BMP since BMP uses the Remote Serial Protocol (RSP).

@sidprice I launch GDB and communicate with that. I do not talk directly to the BMP.

bobbatcomcastdotnet, 2020-08-01:
That being said it is possible that GDB generates different RSP commands when MI2 is used, I don't know. However, whatever RSP commands are being generated the BMP should handle them correctly.

sidprice, 2020-08-01:
@bobbatcomcastdotnet Using MI2 to communicate with GDB has no effect on the communication with the BMP. In fact in the ImageCraft debugger I wrote I also used GDB/MI, worked fine with BMP and ctxLink.

Unfortunately I don't have an STM32H743 to try it out :{

bobbatcomcastdotnet, 2020-08-01:
@sidprice I have a couple more STM32H743 boards on order. Perhaps I could send you one to test on. The application I am working on is a HID composite host that will eventually connect to an Rii 2.4G Mini Wireless Keyboard with Touchpad Mouse. As you might expect there currently is no libopencm3 USB support for the H743 (actually there is almost no H743 support at all from libopencm3) so I am using code that came from ST Micro and uses the HAL library. The code utilizes a host structure to hold the host data and a pointer to that host is passed all over the place. It is that structure I need to examine using the print /x *phost GDB command and it is that command that the BMP fails on with an error message. JLink on the other hand does not fail. I have other code that uses the H743 which also fails to access global vars using the BMP.

sidprice, 2020-08-01:
@bobbatcomcastdotnet I would be happy to try and take a look. @UweBonnes is the expert on the BMP interface to the target, however, maybe I can collect enough data to enable him to see the issue.

If you have a small project that demonstrates the issue you should send that along too.

bobbatcomcastdotnet, 2020-08-01:
@sidprice When I receive the boards (they are from China) I will send one along to you. I will try to work on a small project that demonstrates the problem but that may be difficult since the problem only appears to happen on my larger code :(

UweBonnes, 2020-08-01:
@sidprice: NUCLEO-H7 board are easily available

@ bobbatcomcastdotnet 
It is a pity the Gareth is not around. I remember that he once mentioned MI2. I do not know if that was in the context of RTOS support or wit debugging. If you suspect problems with the GDB protocol, in gdb enable protocoll debugging bu "set debug remote 1" to the the packet exchange.

sidprice, 2020-08-01:
@bobbatcomcastdotnet A while ago @UweBonnes  asked the following:

When you have that problem via 'print /x variable` , resolve the address via "p &variable" and try to dump them memory via "x address"

I did not seen any answer to that ... and I tend to agree with @UweBonnes that this looks like either a GDB issue, or at least GDB having issues with the particular ELF file.

Maybe upload the ELF file and tell us a variable you have issues with.

bobbatcomcastdotnet, 2020-08-01:
@UweBonnes I will try adding "set debug remote 1" to my GDB setup and see what happens.

@sidprice The question @UweBonnes asked:

_When you have that problem via 'print /x variable` , resolve the address via "p &variable" and try to dump them memory via "x address"_

was answered in the original post by:

The error GDB returns is "access to memory location 0xxxxx is denied". **However, if I do a memory dump I can access that memory address without problems.**

_and I tend to agree with @UweBonnes that this looks like either a GDB issue, or at least GDB having issues with the particular ELF file._

If this were a GDB issue why would the jlink probe not show the exact same problem? That is exactly why I wanted to try the jlink probe before I reported this issue.

sidprice, 2020-08-01:
@bobbatcomcastdotnet It is mysterious that BMP treats these variables differently, just struggling to understand why.

I feel there is something missing information-wise, just don't know what that may be, or what questions to ask to elicit the information, sorry.

bobbatcomcastdotnet, 2020-08-01:
@sidprice Nothing to be sorry about at all. At the beginning of my IDE development project I spent a great deal of time studying the RSP protocol and the BMPs implementation of it and yet I too am baffled by this issue.

sidprice, 2020-08-01:
I have ordered nucleo-h743 from Mouser, probably won't get here until Thursday, probably quicker than China :)

bobbatcomcastdotnet, 2020-08-01:
@sidprice _"probably quicker than China :)"_

Anything is quicker than China :)

The only problem I see with this board is the way it is clocked. It does not include a crystal for the HSE oscillator. Instead it uses the 8mHz MCO output from the on-board ST-LINK. That means it will not work with my code as is :(

I have one of these boards so I will try to produce an example app that demonstrates the issue.

bobbatcomcastdotnet, 2020-08-05:
@UweBonnes This is a repeat of a post I sent to Discord. I wanted to post it here to continue with this issue.

"I really don't think the MI interface to GDB in and of itself is the problem I experienced with the H7. But maybe the GDB issues a different set of RSP commands when MI is being used and it is those RSP commands that are not being handled properly. In any case the key to finding the problem lies in determining exactly what RSP packets are being sent to the BMP and what the BMP responds with when the problem occurs. Since it appears that the BMP-hosted application for Windows does not seem to work a different method of collecting this information is necessary. When I was developing my IDE I once had some code that inserted a set of virtual COM ports between GDB and the BMP so I could capture the actual RSP packet communications. Perhaps I can resurrect that code so that we can find out what RSP packets are being passed between GDB and BMP when the problem occurs."

sidprice, 2020-08-05:
@bobbatcomcastdotnet My Nucleo is supposed to arrive today, assuming I can believe FedEx. Do you have a small program to reproduce?

UweBonnes, 2020-08-05:
I have also H7 boards to test, so send to

[30 more replies]

closed 2021-12-16 by esden