blackmagic-debug/blackmagic #705: jtag_scan does not work for OrangeCrab v0.2

hcab14, 2020-07-30:
```
/src/blackmagic -t  -j
INFO: Open USB 0480:0103 failed
Using 1d50:6018 BFC9A3F7 Black Sphere Technologies Black Magic Probe
Remote is Black Magic Probe v1.6.1-532-g71e5b4d
Running in Test Mode
Target voltage: 3.3V Volt
Resetting TAP
Change state to Shift-IR
Scanning out IRs
Return to Run-Test/Idle
Change state to Shift-DR
jtag_scan: Sanity check failed: BYPASS dev count doesn't match IR scan
Given target nummer 1 not available
```

UweBonnes, 2020-07-30:
Bmp does not know about the orange crab until you aquaint BMP to orangecrab.  Add code to jtag_devs.c.

What devices are to be expected in the orangecrab jtag chain?

sidprice, 2020-07-30:
It is a RSC-V I think.

tarikku, 2020-07-31:
OrangeCrab (https://github.com/gregdavill/OrangeCrab) is a dev-board with a Lattice ECP5-25F FPGA, so it could be loaded with almost anything.
It comes with a bootloader (part of the on-board QSPI flash chip), as well as a header for JTAG access.

Which use case are you thinking of, Christoph? E.g. loading a new FPGA 'image' into the onboard flash, debugging...?

hcab14, 2020-07-31:
Thank you all for your replies.

> Which use case are you thinking of, Christoph? E.g. loading a new FPGA 'image' into the onboard flash, debugging...?

I only wanted to do a `JTAG` scan and maybe wrongly assumed that the `jtag_scan` functionality would be independent from the ARM debugging.

BTW: the IR scan finds two devices with IR lengths 2 and 6. If I modify the code not to abort after the DR scan in BYPASS mode where only one device is being found, the `IRCODE` is read out (can be seen when using bit level debugging), however it is not displayed later on.

I am not an expert in JTAG protocol but maybe the behavior above can tell you something meaningful.

Feel free to close this issue.

sidprice, 2020-07-31:
@hcab14 Blackmagic Probe is not a general purpose JTAG device it is specifically designed for Cortex MCUs :(

UweBonnes, 2020-07-31:
The BMP is specifically designed for CortexM, but the implemented JTAG generics can serve for general purpose JTAG. Firmware BMP does not use it in that way, but other programm can use is via the remote protocoll.

esden, 2021-12-16:
We don't currently have plans to support "raw" JTAG. So this issue is not really actionable and thus I am closing it.

closed 2021-12-16 by esden