blackmagic-debug/blackmagic #713: Can't flash to STM32F103C8T6

luqasz, 2020-08-21:
Hi.

I've noticed that bin file is 87KB in size while uC on my F3 discovery and stlink clones have uC with 64KB flash size. Am I compiling it wrong or something changed ?

I've also noticed that some pages ([this](https://wiki.paparazziuav.org/wiki/STLink) and [this](https://medium.com/@paramaggarwal/converting-an-stm32f103-board-to-a-black-magic-probe-c013cf2cc38c)) online confirm BMP image to work on 64KB flash size uC.

UweBonnes, 2020-08-21:
Native STM32F103x8 are produced like the xB with 128 kB flash. There is no guarantee, that this upper flash works, but there have been no negative reports so long. Read more in src/platforms/stlink/Flashsize_F103.

Clone and fake F103 probably will not work.

jspngh, 2020-10-11:
As a note for anyone that comes to this issue because they want to use a cheap blue pill clone with BMP:
some (most?) of these devices report 64K flash, but apparently have 128K.
I found [this blog](https://microcontrollerelectronics.com/how-to-convert-an-stm32f103c8t6-into-a-black-magic-probe/) that shows how it can be done.

In case, the link doesn't work anymore in the future, it basically comes down to this:
```
openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg -c "init" -c "halt" -c "stm32f1x unlock 0" -c "shutdown"
make PROBE_HOST=stlink
st-flash erase
st-flash --flash=0x20000 write blackmagic_dfu.bin 0x8000000
st-flash --flash=0x20000 --reset write blackmagic.bin 0x8002000
```
With the `--flash=0x20000` you can specify how much flash your device has.

UweBonnes, 2020-10-11:
BMP src/platforms/stlink/README.md and src/platforms/stlink/Flashsize_F103 are more actual references.

jspngh, 2020-10-14:
That's indeed some very nice documentation and a better source of information.
Thanks for the hint!