blackmagic-debug/blackmagic #736: Support for LPC55?

jans23, 2020-09-22:
I see blackmagic doesn't support LPC55. Is support for LPC55 planned or do any unofficial patches exist for such?

UweBonnes, 2020-09-22:
BMP is a community effort. Somebody must have some LPC55 board and write and test support for it.

UweBonnes, 2020-12-09:
There is support for LPC54 now in master. Perhaps LPC55  only requires small changes?

esden, 2021-12-16:
@jans23 Do you want to comment on this? All we might need is to add the correct ID to the BMP firmware and your device will be supported.

jans23, 2021-12-16:
Didn't had time to test it yet.

esden, 2021-12-16:
Ok, sounds good. :)

UweBonnes, 2021-12-16:
Please have a look at https://github.com/UweBonnes/blackmagic/tree/lpc55. It  puts detecting the device on more solid ground. I have a MCULINK to test things, but I do not understand the device good enough to add the memory map and I do not feel like I have time to implement the flashing algo.  #849 needs also to be considered.

dragonmux, 2023-01-30:
Support for this is now slated to land in v1.10

PR #1374 (2023-04-02): Feature: Initial LPC55 support

## Detailed description

This PR is a revamp of #861 to meet the current code base standards and contribution guidelines, and replaces that PR. This provides initial basic support for LPC55 targets which will then be supplemented by #1189 which expands this considerably to provide full target support.

This basic support provides detection, "debug mailbox" recovery support, and recovery-mode mass erase.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x [... cut]
closed 2023-04-02 by esden