blackmagic-debug/blackmagic #740: usb icdi error windows 10

daguirrem, 2020-09-30:
[image]
The launchpad-icdi version appears as unknown device on device manager on windows 10 with the latest version of firmware, i have a older version (i don't know exactly which one) that works well

UweBonnes, 2020-09-30:
Is the the same PC setup as with the succeeding tests earlier? Is the blackmagic.inf file installed? Does the icdi  work on a linux box?

daguirrem, 2020-09-30:
Yes, i test the 2 firmwares at same time (and on same pc) and the older works fine, i think that can't probe on a VM becouse Windows alerts me like when we connect a damage usb, directly can't update (or do something) with the driver and the device.

### The older version that works (with the number of firmware):
[image]

### The new version (unknow device) (with windows event report) :
[image]
[image]
[image]

UweBonnes, 2020-09-30:
As asked above, check that driver/blackmagic.inf is installed

daguirrem, 2020-09-30:
is the .inf necesary for windows 10? when i try install appears error: "Third-party INF does not contain digital signature information" and the wiki just specify the driver for windows 7 and older

UweBonnes, 2020-09-30:
Probably not, but I do not work on windows. Compiles some old firmware, load it and check if that works. If you are sure it is an error with the firmware, do a bisect.

esden, 2021-12-16:
As the icdi usb error patch is merged now I assume this issue is resolved? Closing.

closed 2021-12-16 by esden