blackmagic-debug/blackmagic #753: Attaching to Remote target failed

mdarii, 2020-10-15:
I'm trying to connect to NRF52832 device do unlock it, but I'm getting following error
```
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) set non-stop on
(gdb) mon swdp_scan
Target voltage: 3.12V
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) attach 2
Attaching to Remote target
Attaching to Remote target failed with: T05
```
Could somebody help with some suggestions what could be cause of that issue?

I'm following this guide: https://github.com/AndruPol/nrf52832-recover

UweBonnes, 2020-10-15:
Please try w/o "set non-stop on" and report your setup like firmware version.

mdarii, 2020-10-15:
@UweBonnes without "set non-stop on" but looks like device is still locked, I'm not able to upload software using st-link:
```
nRF52 device has a CTRL-AP dedicated to recover the device from AP lock.
A high level adapter (like a ST-Link) you are currently using cannot access
the CTRL-AP so 'nrf52_recover' command will not work.
Do not enable UICR APPROTECT.
```
I'm a bit newbie with that devices may by somebody could help me with working how to unlock and nrf52 module?

UweBonnes, 2020-10-15:
You now did run with something different as I womder where the message about the high level adapter comes from. You also did not run with "set debug remote 1" as requested. That adds confusion!

mdarii, 2020-10-15:
@UweBonnes  sorry missed you message about debug, here I've tried it:
```
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) set debug remote 1
(gdb) mon swdp_scan
Sending packet: $qRcmd,737764705f7363616e#3c...Ack
Packet received: O54617267657420766f6c746167653a20332e3038560a
Target voltage: 3.08V
Packet received: O417661696c61626c6520546172676574733a0a
Available Targets:
Packet received: O4e6f2e20417474204472697665720a
No. Att Driver
Packet received: O20312020202020204e6f72646963206e52463532204d340a
 1      Nordic nRF52 M4
Packet received: O20322020202020204e6f72646963206e524635322041636365737320506f7274200a
 2      Nordic nRF52 Access Port 
Packet received: OK
(gdb) attach 2
Attaching to Remote target
Sending packet: $vAttach;2#38...Ack
Packet received: T05
Packet vAttach (attach) is supported
Sending packet: $qC#b4...Ack
Packet received: 
Sending packet: $Hg0#df...Ack
Packet received: 
Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
Packet received: l
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring
Sending packet: $g#67...Ack
Packet received: b5134431
Sending packet: $pf#d6...Ack
Packet received: cf
Packet p (fetch-register) is supported
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
Sending packet: $qfThreadInfo#bb...Ack
Packet received: 
Sending packet: $qL1200000000000000000#50...Ack
Packet received: 
Sending packet: $qXfer:memory-map:read::0,3fb#e5...Ack
Packet received: m<memory-map></memory-map>
Sending packet: $qXfer:memory-map:read::19,3fb#1f...Ack
Packet received: l
warning: while parsing target memory map (at line 1): Required element <memory> is missing
Sending packet: $m9db86ece,4#67...Ack
Packet received: E01
0x9db86ece in ?? ()
(gdb) mon erase_mass
Sending packet: $qRcmd,65726173655f6d617373#a3...Ack
Packet received: OK
```
but looks like device is still blocked

UweBonnes, 2020-10-15:
Replug and all should be fine!

mdarii, 2020-10-16:
@UweBonnes nope, it is still locked, I'll try unlock other module, may be something wrong in my schematics

UweBonnes, 2020-10-16:
This is how mon s looks like on a protectec device:
(gdb) mon s
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Nordic nRF52 Access Port

mdarii, 2020-10-16:
@UweBonnes that means I have other type of issue, but I'm not able to upload software using st-link I've get that error:
```
nRF52 device has a CTRL-AP dedicated to recover the device from AP lock.
A high level adapter (like a ST-Link) you are currently using cannot access
the CTRL-AP so 'nrf52_recover' command will not work.
Do not enable UICR APPROTECT
```
I assume this means device is locked, I'm wrong?

mdarii, 2020-10-16:
may be something wrong with my visual studio settings, I'll check something

esden, 2021-12-16:
@mdarii any updates on this? Did you manage to find a solution to your issue?

esden, 2022-08-02:
As there was no response here for >6 months I am closing this.

closed 2022-08-02 by esden