blackmagic-debug/blackmagic #754: Blackmagic PC application and Nucleo board (l476rg)  to program Cortex MCU (tested with nRF52)

pshri, 2020-10-16:
Thank you Team for the blackmagic program. 

I was able to build it in a 64 bit Windows 10 machine using [Git for Windows SDK](https://github.com/git-for-windows/build-extra/releases/latest). I had to add [libusb](https://packages.msys2.org/package/mingw-w64-x86_64-libusb) and [libftdi](https://packages.msys2.org/package/mingw-w64-x86_64-libftdi) for a successful build. Attaching [pre-built binaries](https://github.com/blacksphere/blackmagic/files/5388589/blackmagic.zip) with gdb from [GNU Arm Embedded Toolchain](https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2020q2/gcc-arm-none-eabi-9-2020-q2-update-win32.exe?revision=50c95fb2-67ca-4df7-929b-55396266b4a1). Run "fw_update.bat filename" to flash the firmware. Direct programming via blackmagic didn't work hence had to use gdb.

**Debug Connections**
**=============**

Nucleo board (l476rg) <------------->    Supported Cortex MCU board
        CN4-1 <------------------------------>   Vref
        CN4-2 <------------------------------>   SWDCLK
        CN4-3 <------------------------------>   GND
        CN4-4 <------------------------------>   SWDIO

Sample output with nRF52840
blackmagic.exe -t

INFO: Open USB 01d9:2007 failed
Using 1234:5678 0226AA525436666547143849 STMicroelectronics STM32 STLink
V2J37M26
Leaving DEBUG Mode
Running in Test Mode
Target voltage: 3.26V Volt
Leaving MASS Mode
Enter SWD
Read Core ID: 0x2ba01477
RESET_SEQ failed
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040
ROM: Table BASE=0xe00ff000 SYSMEM=0x1, PIDR 0x02002c4008
0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0x04000bb00c  DEVTYPE = 0x00 ARCHID = 0x0000)-> cortexm_probe
1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04003bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003  DEVTYPE = 0x00 ARCHID = 0x0000)
3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04003bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
4 0xe0040000: Debug component - Cortex-M4 TPIU (Trace Port Interface Unit) (PIDR = 0x04000bb9a1  DEVTYPE = 0x11 ARCHID = 0x0000)
5 0xe0041000: Debug component - Cortex-M4 ETM (Embedded Trace) (PIDR = 0x04000bb925  DEVTYPE = 0x13 ARCHID = 0x0000)
ROM: Table END
AP   1: IDR=02880000 CFG=00000000 BASE=00000000 CSW=00000000
***  1      Nordic nRF52 M4
***  2      Nordic nRF52 Access Port
Ram   Start: 0x20000000, length   0x40000
Flash Start: 0x10001000, length    0x1000, blocksize   0x1000
Flash Start: 0x00000000, length  0x100000, blocksize   0x1000

UweBonnes, 2020-10-16:
So the real issue for you is : "Direct programming via blackmagic didn't work ..."
Show the command line, add -v3 and show the output. Remember, direct programming uses a bin file. W/o other arguments is starts at the lowest flash, 0 for the NRF.

pshri, 2020-10-16:
Actually I am quite happy that blackmagic allows me to program other vendor MCU using nucleo board. Just wanted to share it with the community.

blackmagic.exe -v3 -a 0x0 test.bin

INFO: Open USB 01d9:2007 failed
Using 1234:5678 0226AA525436666547143849 STMicroelectronics STM32 STLink
V2J37M26
Leaving DEBUG Mode
Target voltage: 3.26V Volt
Leaving MASS Mode
Enter SWD
Read Core ID: 0x2ba01477
RESET_SEQ failed
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040
ROM: Table BASE=0xe00ff000 SYSMEM=0x1, PIDR 0x02002c4008
0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0x04000bb00c  DEVTYPE = 0x00 ARCHID = 0x0000)-> cortexm_probe
1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04003bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003  DEVTYPE = 0x00 ARCHID = 0x0000)
3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04003bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
4 0xe0040000: Debug component - Cortex-M4 TPIU (Trace Port Interface Unit) (PIDR = 0x04000bb9a1  DEVTYPE = 0x11 ARCHID = 0x0000)
5 0xe0041000: Debug component - Cortex-M4 ETM (Embedded Trace) (PIDR = 0x04000bb925  DEVTYPE = 0x13 ARCHID = 0x0000)
ROM: Table END
AP   1: IDR=02880000 CFG=00000000 BASE=00000000 CSW=00000000
***  1      Nordic nRF52 M4
***  2      Nordic nRF52 Access Port
Erase    1048576 bytes at 0x00000000
STLINK_SWD_AP_STICKY_ERROR
Erased failed!

esden, 2021-12-16:
Considering that #873 is closed I assume this issue here is also resolved? If that is not the case then please feel free to reopen this issue.

closed 2021-12-16 by esden

pshri, 2022-02-04:
imho 873 is not related. Sorry I don't have the board anymore to test this again

esden, 2022-02-04:
No problem. Thanks for the information.