blackmagic-debug/blackmagic #759: Support EFR32BG2x Series 2

tyler123durden, 2020-10-18:
Hello, 

We were using black magic quite for quit a time to flash Silabs EFR32 Series 1 chips but now we are switching to series 2 with M33 cores and they are not supported in bmp. 
So I tried to look in the sources and it seems M33 support should at least work for some STM chips? 
I'm able to read the ROM table but reading flash content or the device information registers fail. So for now I have no clue  where to go from here or where I could get further information to try to get it working. 

The device scan also shows the second RF controller a M0+ and I can access the flash contents of that device. 

So maybe anyone could give a brief overview on cortex m33 support and its current state in bmp and what is missing.

UweBonnes, 2020-10-19:
What is similar to the other CortexM shopuld be supported now. Git head forgets to restart the device after halting it for romtable scan, and so debugging goes astray. Restarting a not yet supported CortexM lets me e.g. attach to a STM32L552 and  single step it. Please try https://github.com/UweBonnes/blackmagic/tree/fixes where this is already added.

However CortexMx3 has more security features. Somebody with (deep )insight into these feature has to implement what else is needed.

I hope you start to look deeper in the BMP code. If you do, I suggest you use bmp-hosted, as no additional flashing step is needed after changes for debugging and debugging output appears on the starting console without additional further setup. Feel free to join the BMP channel on discord: https://1bitsquared.com/pages/chat

tyler123durden, 2020-10-19:
Thanks for the quick reply. At least I've made some progress. 
With the hosted version I can read the flash contents of the EFR32BG22 and could find the correct device ID in the device information page. So the issue with the native version is maybe some timing stuff, no sure how the hosted version influences the behaviour of code. 
Your tree with the fixed version didn't help, I also tried your restarting fix for silabs the device with no luck. 
But for now that is a good starting point for further debugging.

UweBonnes, 2020-10-19:
Be more verbose, e.g. What probe, native, other firmware or ftdi/stlink/cmsis/jlink? Any error messages with hosted? Post them if they are not obvious to you.

UweBonnes, 2020-10-19:
Otherwise the fixes tree should allow to open a debu session also on not recognized CorteM parts.

tyler123durden, 2020-10-19:
Sorry for the missing details. I'm using native black magic probe I think version 2.1c. 
Firmware on the probe is from the fixes tree. I made some changes in the erf32 target source to decode the new device ID from silabs device information page. But that didn't work out, I was not able read from Flash with the target_mem_read32() function the return value was always 0 for all Flash areas I tried. I also added target_halt_resume(t, 0) from the STM case to try to restart the core as you mentioned. 
Then I tried the hosted version in an Linux environment with the same code and that version read the Flash areas as expected. 

So here is the console output from the hosted version where the reads worked as expected. 
My debug outputs are the lines between, I didn't modifiy the code at other places

cortexm_probe
...
SI: part:  128 version: 2

At the native version only the reads from the ROM table (e00ff000 -e000ed08 ) worked. 
The other reads returned 0 for the M33 core. For the M0+ core all reads were successful at native and hosted version, but of course the addresses used here for device information und user data page are kind of garbage I didn't check them for the M0+ I wanted only to test for the M33, but the flash programm area reads seemed to be valid for the M0+. 

```
$ sudo ./blackmagic -t
Using 1d50:6018 C1E7D0DD Black Sphere Technologies Black Magic Probe
Remote is Black Magic Probe v1.6.1-618-g513e865-dirty
Running in Test Mode
Target voltage: 3.0V Volt
RESET_SEQ failed
TARGETID 02d416e7
AP   0: IDR=84770001 CFG=00000000 BASE=e00fe003 CSW=03800040
ROM: Table BASE=0xe00fe000 SYSMEM=0x00000001, designer 673 Partno d41
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c9
 0 0xe000e000: Debug component - Cortex-M33 (System Control Space) (PIDR = 0x04000bbd21  DEVTYPE = 0x00 ARCHID = 0x2a04) -> cortexm_probe
Silabs 6ba02477
Device Information Page: 
fe08000 : 706ca56
fe08004 : 21608b0
fe08008 : 1000103
User data UD
fe00000 : 3704
fe00004 : 1
fe00008 : 0
e000ed00 : 410fd214
e000ed04 : 400000
e000ed08 : 0
Try read flash
0 : 20002fe8
4 : 8099
8 : 108a1
c : 80c5
10 : 80c5
e00ff000 : fff0f003
e00ff004 : fff02003
e00ff008 : fff03003
SI: part:  128 version: 2
Dev idx a Version: 2
 1 0xe0001000: Debug component - Cortex-M33 (Data Watchpoint and Trace) (PIDR = 0x04000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a02)
 2 0xe0002000: Debug component - Cortex-M33 (Breakpoint Unit) (PIDR = 0x04000bbd21  DEVTYPE = 0x00 ARCHID = 0x1a03)
 3 0xe0000000: Debug component - Cortex-M33 (Instrumentation Trace Macrocell) (PIDR = 0x04000bbd21  DEVTYPE = 0x43 ARCHID = 0x1a01)
 4 Entry 0xfff41002 -> Not present
 5 0xe0041000: Debug component - Cortex-M33 (Embedded Trace) (PIDR = 0x04002bbd21  DEVTYPE = 0x13 ARCHID = 0x4a13)
 6 0xe0042000: Debug component - Cortex-M33 (Cross Trigger) (PIDR = 0x04000bbd21  DEVTYPE = 0x14 ARCHID = 0x1a14)
 7 Entry 0xfff44002 -> Not present
 ROM: Table END
1 0xe0040000: Debug component - Cortex-M33 (Trace Port Interface Unit) (PIDR = 0x04000bbd21  DEVTYPE = 0x11 ARCHID = 0x0000)
2 0xe00fd000: PrimeCell peripheral - System TSGEN (Time Stamp Generator) (PIDR = 0x04001bb101  DEVTYPE = 0x00 ARCHID = 0x0000)
ROM: Table END
AP   1: IDR=54770002 CFG=00000000 BASE=00000003 CSW=00000040
AP   3: IDR=84770001 CFG=00000000 BASE=f0000003 CSW=43800040
ROM: Table BASE=0xf0000000 SYSMEM=0x00000001, designer 673 Partno 52d
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008  DEVTYPE = 0x00 ARCHID = 0x0000) -> cortexm_probe
Silabs 6ba02477
Device Information Page: 
fe08000 : 0
fe08004 : 0
fe08008 : 0
User data UD
fe00000 : 0
fe00004 : 0
fe00008 : 0
e000ed00 : 410cc601
e000ed04 : 0
e000ed08 : 0
Try read flash
0 : 4000
4 : 98d
8 : 331
c : 341
10 : 351
e00ff000 : fff0f003
e00ff004 : fff02003
e00ff008 : fff03003
SI: part:  0 version: 2
Dev idx 270f Version: 2
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM: Table END
 0xf0001000: Debug component - Unknown (PIDR = 0x04000bb9a6 DEVTYPE = 0x14 ARCHID = 0x1a14)
ROM: Table END
***  1      SERIESV20 65535 F65535 EFR32 Version2 M33
*** 2 Unknown ARM Cortex-M Designer 673 Partno 52d M0+
Ram   Start: 0x20000000, length 0x3fffc00
Flash Start: 0x0fe00000, length     0x400, blocksize    0x52c
Flash Start: 0x00000000, length 0x3fffc00, blocksize    0x52c
```
In the next days I'll also try to get a log with the native version where the reads returned 0 and you can compare them but as far as I remember the log was identical beside the memory content.

UweBonnes, 2020-10-20:
This all looks fine to me. Designer 673 is Energy micro, the SOC has an additional M0 at AP3. Blocksize "0x57c" seems odd. A short peak at the datasheet lets me think it is 8k. I can not judge if the printout of the devices information page make sense, maybe the location is somewhere else or it has no or another implementation. Attaching and debugging on both CPUs should be possible. If not, post error messages from gdb and the starting console for hosted here. Somebody with the part needs to look if the programming algo is similar to the other EF parts and needs to adapt or eventually add a new target.

tyler123durden, 2020-10-20:
The layout and address of the device information page has changed at series 2 chips. The read part ID 128 from the new device info location is correct and I mapped it already to "SERIESV20" we must further examine the device information table to get correct page size and flash size but this should be straight forward.

UweBonnes, 2020-11-27:
Awaiting a PR.

esden, 2021-12-16:
@tyler123durden do we have any updates on this? Are you still working on adding support for this part?

esden, 2022-08-02:
As there was no reply in >6 months I am assuming there is no work being done on this. I am closing it for now. Feel free to reopen this issue or open a new one when work resumes.

closed 2022-08-02 by esden

lucasromeiro, 2022-09-06:
@tyler123durden Hi, how are you?  

I bought an EFR32BG22 and I need to program it. Could you tell me how you solved your problem? A Segger recorder is very expensive. I didn't find another alternative.