blackmagic-debug/blackmagic #760: Is there a way to write a binary to flash at specific address?

pvonmoradi, 2020-10-21:
Thanks for this awesome project!   
I'm trying to modify a makefile which assumes st-link for flashing the binary to the target.  
```
flash:	$(BINARY).bin
	# $(STFLASH) $(FLASHSIZE) write $(BINARY).bin 0x8000000
	# openocd -f interface/ft232r.cfg -f board/stm32f103c8_blue_pill.cfg\
	#	 -c "program $(BINARY).bin verify reset exit 0x08000000"
	arm-none-eabi-gdb -nx --batch \
		-ex 'target extended-remote /dev/ttyBmpGdb' \
		-ex 'monitor swdp_scan' \
		-ex 'attach 1' \
		-ex 'load' \
		-ex 'compare-sections' \
		-ex 'kill' \
		$(BINARY).elf
```  
With openOCD, I can directly flash the binary to 0x08000000. How can I do the same with BMP? (The makefile creates the elf file but I want to remain consistent).
https://github.com/blacksphere/blackmagic/wiki/GDB-Automation

UweBonnes, 2020-10-21:
Compile for hosted "make PROBE_HOST= hosted" and run "blackmagic(.exe) <binary>.bin"

pvonmoradi, 2020-10-21:
@UweBonnes Thanks. I guess that's possible but it does not use the gdb loaded on the programmer.  
This seems to be a bug with GDB itself:  https://gitter.im/blacksphere/blackmagic?at=5b5caedc12f1be713760f001

> Piotr Esden-Tempski
> @esden
> Jul 30 2018 00:19
> I think restore does not support flashing. This is a "bug" in gdb... for some reason the person that implemented flashing did not implement it for restore. :(
> you will have to take the binary and use objcopy to convert it to a hex file and then use load to flash the binary.
> (I will have to look but I think there is a thread on the gdb mailinglist regarding the restore issue)
@esden Any updates on this?

UweBonnes, 2020-11-27:
Closing this issue as there is a workaround and no culprit on the BMP side

closed 2020-11-27 by UweBonnes