blackmagic-debug/blackmagic #765: Can we program TI CC1310 with BMP?

chansdad, 2020-10-28:
Hello folks , 

Wanted to know if anyone programmed cc1310 with BMP? if so , is it out of the box or do i need to upgrade firmware ? 

Appreciate your response.

UweBonnes, 2020-10-28:
Nobody provided  programming algorithme up to now. You are welcome to contribute. Debugging as CortexM should be available. If you run the part against a recent BMP, designer and partno will be printed. I appreciate  if you let me knoe this data.

chansdad, 2020-10-28:
Wish I can , unfortunately not an expert , i am still trying to understand which programmer i should use and how to load the program on cc1310 boards .. exploring what is out there at the present time.

UweBonnes, 2020-10-28:
OpenOCD knows about the CC13x0.  If you dare, try #570 to get a dedicated BMP firmware running with OpenOCD.